offspot/kiwix-hotspot #563: Fails to retrieve zim file

Popolechien, 2021-04-13:
Job [L462R6074](https://cardshop.hotspot.kiwix.org/orders/L462R6074) crashed for the following reason: `Error downloading http://mirror.download.kiwix.org/zim/wikisource/wikisource_en_all_maxi_2021-03.zim to /data/kiwix-hotspot.cache/wikisource_en_all_maxi.en.zim`

This seems to be a recurring issue lately - any problem with syncing up with the library?

rgaudin, 2021-04-13:
No, as you can see this file exists and download started. aria2c exited with a status-code of `1` which means `unknown error` (it has many many different error codes for most scenarios like lack of space or timeout, etc).

Nothing we can do about this unless this reproduces so closing it for now.

I launched this exact download and will wait till it completes to exclude the possibility of this particular file being inconsistent.

Popolechien, 2021-04-13:
re-opening as it re-occured.
https://cardshop.hotspot.kiwix.org/orders/L464R6075

rgaudin, 2021-04-13:
Arg. and it failed on the very same file.

kelson42, 2021-04-19:
@Popolechien Now that I have cleaned the disk of all errors, would that be hard to retry again?

Popolechien, 2021-04-19:
@kelson42 restarted [L499R607D](https://cardshop.hotspot.kiwix.org/orders/L499R607D) : seems to be working

Popolechien, 2021-04-20:
Reopening as we had the exact same issue again last night with [L500R607D](https://cardshop.hotspot.kiwix.org/orders/L500R607D) and [L501R607D](https://cardshop.hotspot.kiwix.org/orders/L501R607D) but we two different files:

This time we get 
```
Error downloading http://mirror.download.kiwix.org/zim/gutenberg/gutenberg_en_all_2021-03.zim to /data/kiwix-hotspot.cache/gutenberg_en_all.en.zim
aria2c returned 1
```
and 
```
Error downloading http://mirror.download.kiwix.org/zim/wikipedia/wikipedia_en_all_maxi_2021-03.zim to /data/kiwix-hotspot.cache/wikipedia_en_all_maxi.en.zim
aria2c returned 1
```

kelson42, 2021-04-20:
@rgaudin It is not possible to have more debug  infos from aria2c?

Popolechien, 2021-04-20:
I relaunched a creation with one of the two configs (gutenberg_en_all) and it worked no problem (=not reproducible).

rgaudin, 2021-04-20:
> @rgaudin It is not possible to have more debug infos from aria2c?

Not really, as you can [see here](https://aria2.github.io/manual/en/html/aria2c.html#exit-status), aria2c is pretty precise about issues based on status code.

I have no idea if there is more info to it in stdout/stderr as I have not been able to reproduce it. We can't just display the output on the hotspot as very rely on specific params to tweaks its output so we can parse progress.

rgaudin, 2021-04-20:
I'm looking into updating aria2c in hotspot as we currently bundle version 1.33 while 1.35 is available. Going through the changelog…

Popolechien, 2021-04-24:
Reopening because of https://cardshop.hotspot.kiwix.org/orders/L517R6083
Crashed after downloading about a third of Gutenberg_2021_03.zim

rgaudin, 2021-04-30:
Seems obsolete now especially with the use of the RPN. Let's close it until we face it again.

commit 3c69dc16 (2021-04-21): Fixed #563: updated aria2c to 1.35.0

Not certain it will fix it for good but that's our best chance with what we know.
closed 2021-04-30 by rgaudin