offspot/kiwix-hotspot #565: Kiwix Hotspot ZIM content unresponsive after a few minutes on Raspberry Pi

mrcage, 2021-04-15:
I am experiencing issues with Kiwix Hotspot images on Raspberry Pi (Model 4B and 3B+ tested so far using Samsung microSD cards). Whenever I flash a micro SD card, put it into the Raspberry, start it up and access the Kiwix Hotspot web page. everything works fine for a few minutes, but after that all ZIM-based content (Wikipedias etc) become non-responsive. I can click on any link on the start page, and the browser just tries to load the site for an indefinite period of time. Khan Academy still works. I currently have no way to debug it, as SS access seems to be disabled, and there is no admin mode (even though I set an admin name and password in the Hotspot generator, this is probably a separate issue). I tried both with an image created using the hotspot installer, and an image created using your card shop, as I was suspecting there might be something wrong with the hotspot installer. 

I've also tried several different SD cards from various brands, the issue persists. FYI my image is more than 200GB, I've also tried with a 100GB-ish image before. SSH to the raspberry Pi does not work, and there is no admin panel available in Kiwix (although the images were created with an admin user and password set). 

It seems to be related to the Kiwix Software itself which runs on port 8002. KA Lite which runs on port 8008 is still working fine, but any HTTP request to port 8002 takes indefinitely.

mrcage, 2021-04-15:
Any hints how I could debug this further would be greatly appreciated.

rgaudin, 2021-04-16:
Thanks @mrcage, that really looks like a [kiwix-serve](https://github.com/kiwix/kiwix-tools/) issue indeed. Did you use the full-text search feature when testing this? I know that there has been issues with full-text search on large library that could render kiwix-serve unresponsive. 

Regarding SSH, yes it is disabled. If you can connect the Pi to a screen, you can login with default credentials (`pi`/`raspberry`) and `sudo systemctl start ssh` or `sudo systemctl enable ssh`

mrcage, 2021-04-16:
I was not issuing any search requests, just browsing through some of the wikis and ted talks randomly for testing.

mrcage, 2021-04-16:
I managed to connect via SSH as described. Here is my kiwix-serve command line:

    /usr/local/bin/kiwix-serve --port=8002 /data/packages/aimhi_english.en.zim /data/packages/aimhi_geography.en.zim /data/packages/christianity.stackexchange.com_en_all.en.zim /data/packages/covid19relay_en_all_maxi.en.zim /data/packages/education-et-numerique_fr_all.fr.zim /data/packages/ekopedia_fr_all.fr.zim /data/packages/gutenberg_ar_all.ar.zim /data/packages/gutenberg_fa_all.fa.zim /data/packages/gutenberg_fr_all.fr.zim /data/packages/islam.stackexchange.com_en_all.en.zim /data/packages/jaimelire_fr.fr.zim /data/packages/khan-academy-videos_fr_bases-de-la-geometrie.fr.zim /data/packages/khan-academy-videos_fr_bases-de-l-algebre.fr.zim /data/packages/khan-academy-videos_fr_l-ecologie.fr.zim /data/packages/khan-academy-videos_fr_les-systemes-du-corps-humain.fr.zim /data/packages/khan-academy-videos_fr_sante-et-medecine.fr.zim /data/packages/khan-academy-videos_tr_bilgisayar-proglamlama-bilgisayar-dunyasi.tr.zim /data/packages/khan-academy-videos_tr_elektrik-muhendisligi-fen-bilimleri.tr.zim /data/packages/khan-academy-videos_tr_felsefe-sanat-ve-sosyal-bilimler.tr.zim /data/packages/khan-academy-videos_tr_isik-ve-renkler.tr.zim /data/packages/khan-academy-videos_tr_kimya-hizlandirilmis-kurs-crash-course.tr.zim /data/packages/khan-academy-videos_tr_lineer-cebir-vektorler-uzay-ve-matris-donusumleri.tr.zim /data/packages/khan-academy-videos_tr_mikroekonomi-ekonomi-ve-finans.tr.zim /data/packages/khan-academy-videos_tr_muzik-sosyal-bilimler-ve-sanat.tr.zim /data/packages/lifehacks.stackexchange.com_en_all.en.zim /data/packages/phzh_core-arabic-one-test-zim_ar.en.zim /data/packages/phzh_core-english-one_en.en.zim /data/packages/phzh_core-greek-one_el.en.zim /data/packages/poesies_fr.fr.zim /data/packages/ted_en_playlist-10-talks-by-women-that-everyon.en.zim /data/packages/ted_en_playlist-11-must-see-ted-talks.en.zim /data/packages/ted_en_playlist-how-to-invest-in-social-good.en.zim /data/packages/ted_en_playlist-immigrants-who-founded-amazing-companies.en.zim /data/packages/ted_en_playlist-kickstart-your-creativity.en.zim /data/packages/ted_en_playlist-music-around-the-world.en.zim /data/packages/ted_en_playlist-natural-wonder.en.zim /data/packages/ted_en_playlist-powerful-art-activism.en.zim /data/packages/ted_en_playlist-powerful-poetry-by-fierce-women.en.zim /data/packages/ted_en_playlist-talks-that-explain-difficult-t.en.zim /data/packages/ted_en_playlist-talks-to-restore-your-faith-in-1.en.zim /data/packages/ted_en_playlist-talks-to-watch-when-you-have-a.en.zim /data/packages/ted_en_playlist-ted-talks-by-strong-women-lead.en.zim /data/packages/ted_en_playlist-the-10-most-popular-tedx-talks.en.zim /data/packages/ted_en_playlist-the-beauty-of-strangers.en.zim /data/packages/ted_en_playlist-the-future-of-medicine.en.zim /data/packages/ted_en_playlist-the-love-of-lifelong-learning.en.zim /data/packages/ted_en_playlist-the-most-popular-talks-of-2019.en.zim /data/packages/ted_en_playlist-the-path-to-reconciliation.en.zim /data/packages/ted_en_playlist-the-power-of-fiction-1.en.zim /data/packages/ted_en_playlist-the-power-of-protest.en.zim /data/packages/ted_en_playlist-the-power-of-social-media.en.zim /data/packages/ted_en_playlist-the-rise-of-africa-s-next-gene.en.zim /data/packages/ted_en_playlist-the-road-to-peace.en.zim /data/packages/ted_en_playlist-virtuosos.en.zim /data/packages/ted_en_playlist-what-comes-after-war.en.zim /data/packages/ted_en_playlist-what-does-the-future-look-like.en.zim /data/packages/ted_en_playlist-what-is-home.en.zim /data/packages/ted_en_playlist-what-makes-you-happy.en.zim /data/packages/ted_en_playlist-what-you-need-to-know-about-me.en.zim /data/packages/ted_en_playlist-what-your-data-reveals-about-y.en.zim /data/packages/ted_en_playlist-when-your-beliefs-are-being-ch.en.zim /data/packages/ted_en_playlist-where-do-ideas-come-from.en.zim /data/packages/ted_en_playlist-why-climate-change-is-a-human.en.zim /data/packages/ted_en_playlist-why-do-we-tell-stories.en.zim /data/packages/ted_en_playlist-why-we-need-brave-journalists.en.zim /data/packages/wikem_en_all_maxi.en.zim /data/packages/wikipedia_ar_all_maxi.ar.zim /data/packages/wikipedia_ar_chemistry_maxi.ar.zim /data/packages/wikipedia_ar_computer_maxi.ar.zim /data/packages/wikipedia_ar_football_maxi.ar.zim /data/packages/wikipedia_ar_geography_maxi.ar.zim /data/packages/wikipedia_ar_history_maxi.ar.zim /data/packages/wikipedia_ar_mathematics_maxi.ar.zim /data/packages/wikipedia_ar_medicine_nopic.ar.zim /data/packages/wikipedia_ar_molcell_maxi.ar.zim /data/packages/wikipedia_ar_movies_maxi.ar.zim /data/packages/wikipedia_ar_physics_maxi.ar.zim /data/packages/wikipedia_en_100.en.zim /data/packages/wikipedia_en_astronomy_maxi.en.zim /data/packages/wikipedia_en_chemistry_maxi.en.zim /data/packages/wikipedia_en_climate_change_maxi.en.zim /data/packages/wikipedia_en_computer_maxi.en.zim /data/packages/wikipedia_en_football_maxi.en.zim /data/packages/wikipedia_en_for-schools.en.zim /data/packages/wikipedia_en_geography_maxi.en.zim /data/packages/wikipedia_en_history_maxi.en.zim /data/packages/wikipedia_en_mathematics_maxi.en.zim /data/packages/wikipedia_en_medicine.en.zim /data/packages/wikipedia_en_movies_maxi.en.zim /data/packages/wikipedia_en_physics_maxi.en.zim /data/packages/wikipedia_en_simple_all_maxi.en.zim /data/packages/wikipedia_en_sociology_maxi.en.zim /data/packages/wikipedia_en_top_maxi.en.zim /data/packages/wikipedia_fa_all_maxi [... cut ...] r.zim /data/packages/wikipedia_tr_movies_maxi.tr.zim /data/packages/wikipedia_tr_physics_maxi.tr.zim /data/packages/wikipedia_ur_all_maxi.ur.zim /data/packages/wikipedia_ur_medicine_maxi.ur.zim /data/packages/wikivoyage_en_all_maxi.en.zim /data/packages/wikivoyage_en_europe_maxi.en.zim /data/packages/wikivoyage_fa_all_maxi.fa.zim /data/packages/wikivoyage_fr_all_maxi.fr.zim /data/packages/wiktionary_ar_all_maxi.ar.zim /data/packages/youscribe_fr_primaire.fr.zim

The kiwix-server version is 3.1.2

mrcage, 2021-04-16:
htop shows a huge amount of CPU utilization. This happens after a few minutes of browsing around on kiwix-server UI. At the beginning, utilization is very low, then it suddenly explodes....

[image]

[image]

rgaudin, 2021-04-16:
Could you share the list of packages you used? In a text format so we can reproduce it