offspot/kiwix-hotspot #566: Library is been improperly updated

Popolechien, 2021-04-16:
Order [L475R6079](https://cardshop.hotspot.kiwix.org/orders/L475R6079) failed because of the absence of psiram_de. The file is, in fact, [entirely missing from `download`](http://download.kiwix.org/zim/psiram/) and it has [failed ](https://farm.openzim.org/recipes/psiram_de) consistently to run on the zimfarm for the past 10 months. 

Obviously it should not have been listed in the installer library, but why has it?

kelson42, 2021-04-17:
The problem is that i manually have deleted psiram zim files because they are broken. With the CMS, the library will be updated automatically. But what is the proper handling in the kiwix-hotspot installer when something has been configured but is no available anymore in the library?

kelson42, 2021-04-20:
To me kind of duplicate of But there is a few scenarios where this might be legit, see tickets like https://github.com/kiwix/kiwix-hotspot/issues/571 or https://github.com/kiwix/cardshop/issues/242 or https://github.com/openzim/cms/issues/7.

closed 2021-04-20 by kelson42