offspot/kiwix-hotspot #567: Fails to build with Command returned non-zero exit status 255

Popolechien, 2021-04-18:
From https://cardshop.hotspot.kiwix.org/orders/L482R607B
we got the following worker log:
`Command '[PosixPath('/usr/bin/kiwix-hotspot'), 'cache', '--build', PosixPath('/data'), 'reset']' returned non-zero exit status 255`

Possible reason could be the server crash on the side of download.kiwix.org, however the message is a little cryptic to say the least.

kelson42, 2021-04-20:
@rgaudin Any idea what happens here? We need at least to triage this before the 2.4 release.

rgaudin, 2021-04-22:
I have not been able to reproduce nor find the root cause. I suppose it could be due to the library off time (as this checks the library on startup). Closing until this surfaces again.

closed 2021-04-22 by rgaudin