offspot/kiwix-hotspot #576: Relocate master images

kelson42, 2021-04-21:
Current master images are available at https://download.kiwix.org/hotspots/. This directory will disappear as our plan is to store kiwix hotspot images in S3. But we still need to provide the base image, probably in download.kiwix.org. I would propose to relocated the master images in the respective kiwix-hotspot releases dedicated directory.

rgaudin, 2021-04-22:
As discussed, master images to be relocated from `/hotspots/base/` to `/release/kiwix-hotspot/master-images`.

I have already moved latest image to `/release/kiwix-hotspot/master-images/2021-04-20/`. I think we should only move the previous one which we've been using since version `2.1.0` on  Aug 21, 2019.

@kelson42 please create a redirection from `/hotspots/base/hotspot-master_2019-08-24.img.zip` to `/release/kiwix-hotspot/master-images/2019-08-24/hotspot-master_2019-08-24.img.zip` (file already exists) and remove the whole base folder if you wish to.

kelson42, 2021-04-26:
I have put new images at https://download.kiwix.org/release/kiwix-hotspot/master/

rgaudin, 2021-04-26:
I see redirections are OK ; Should we close?

kelson42, 2021-04-26:
@rgaudin AFAIK:
* Redirections are not OK at all (I don't have made anything)
* Kiwix-hotspot CI is not fixed to upload at the right place

rgaudin, 2021-04-26:
Ah so you left both files?

kelson42, 2021-04-26:
@rgaudin I let you fix the CI. When it is done, we can remove the old files from download.kiwix.org

kelson42, 2021-05-01:
And fix “hostpot”

rgaudin, 2021-05-01:
> And fix “hostpot”

This doesn't need to be fixed. The workflow generating it has the right name. the typo was introduced by manual manipulations.

https://github.com/kiwix/kiwix-hotspot/blob/master/.github/workflows/create-master.yml

kelson42, 2021-05-01:
Why I have seen 3 typos in the code then?

rgaudin, 2021-05-01:
Because the code (japon) uses this manually edited master.

Next master won't have the typo ans selecting a master to use is a matter of copy-pasting its file name and metadata in the json file.

commit e99bcf3d (2021-05-03): Fixed #576: upload master to new location

Create master now have two different modes:
- `release` that uploads a properly named file to the release master folder only on `master-v*` tags
- `CI` that uploads a file with commit ID inside to the CI/hotspot-master folder for all
other cases
commit 3b129cde (2021-05-03): Fixed #576: upload master to new location

Create master now have two different modes:
- `release` that uploads a properly named file to the release master folder only on `master-v*` tags
- `CI` that uploads a file with commit ID inside to the CI/hotspot-master folder for all
other cases
closed 2021-05-03 by rgaudin