offspot/kiwix-hotspot #578: Installation failed multiple times

BenxProductionz, 2021-04-24:
My Installation failed three times so far, it´s always failing with: 

> Installation failed: ssh command failed with status 2. cmd: sudo sh -c 'cd /var/lib/ansible/local && /usr/local/bin/ansible-playbook -vvv --inventory hosts --tags rename,reconfigure --extra-vars="@/var/lib/ansible/local/extra_vars.json" main.yml'
[image]

BenxProductionz, 2021-04-24:
Okay, now I tried to use Version 2.4.0 and now it keeps displaying another error: 

> [2:53:33 PM] Installation failed: wait signal timeout: b'login: '

and I also have the entire log for you attached.
[kiwix log.txt](https://github.com/kiwix/kiwix-hotspot/files/6369627/kiwix.log.txt)

rgaudin, 2021-04-26:
Thank you @BenxProductionz, I've identified the problem. It's a new bug introduced in 2.4.0 by using a more recent version of qemu. We'll have a new release soon.

commit 971cebed (2021-04-26): Fixed #578: shrunk image rounder to next power of 2
commit 6755446b (2021-04-26): Fixed #578: shrunk image rounder to next power of 2
closed 2021-04-26 by rgaudin