offspot/kiwix-hotspot #583: Add selfhosted games

Popolechien, 2021-04-27:
The recent discussion with the advisory board highlighted the fact that a mix of entertainment and education was actually helpful in encouraging buy-in and adoption from learners. 

There is a list of freely licensed games at https://github.com/awesome-selfhosted/awesome-selfhosted#games

[Freeciv](https://github.com/freeciv, [Freedoom](http://freedoom.github.io/) and [Wesnoth](https://wesnoth.org/) are other possible candidates.

Might be worth introducing a few with the installer / cardshop?

rgaudin, 2021-04-27:
Each of them requires work to be added ; your link is just a list referencing different projects. So First step would be to test and identify those you want to add otherwise this is not actionable.

Popolechien, 2021-04-27:
Yes, ultimately a ticket will have to be opened for each we want to include. At this stage it is more of a feasibility question (and a place to list links so we don't need to look for them every time).