offspot/kiwix-hotspot #584: 2.4.2 dies at the end of the process on my ubuntu 20.04

kelson42, 2021-04-28:
```
[15:48:55] TASK [resize : Create test directory] ******************************************
[15:48:55] --> resize : Create test directory
[15:48:55] [STAGE 5/5: Post-process contents (virtualized) - 85%]
[15:48:55] task path: /var/lib/ansible/local/roles/resize/tasks/main.yml:140
[15:48:56] <localhost> ESTABLISH LOCAL CONNECTION FOR USER: root
[15:48:56] <localhost> EXEC /bin/sh -c 'echo ~root && sleep 0'
[15:48:56] <localhost> EXEC /bin/sh -c '( umask 77 && mkdir -p "` echo /var/tmp/ansible-tmp-1619617736.3666978-67585791963378 `" && echo ansible-tmp-1619617736.3666978-67585791963378="` echo /var/tmp/ansible-tmp-1619617736.3666978-67585791963378 `" ) && sleep 0'
[15:48:56] Using module file /usr/lib/python3/dist-packages/ansible/modules/files/file.py
[15:48:56] <localhost> PUT /root/.ansible/tmp/ansible-local-2440jl5kvucd/tmpgpqa1ces TO /var/tmp/ansible-tmp-1619617736.3666978-67585791963378/AnsiballZ_file.py
[15:48:57] <localhost> EXEC /bin/sh -c 'setfacl -m u:user:r-x /var/tmp/ansible-tmp-1619617736.3666978-67585791963378/ /var/tmp/ansible-tmp-1619617736.3666978-67585791963378/AnsiballZ_file.py && sleep 0'
[15:48:57] <localhost> EXEC /bin/sh -c 'chmod u+x /var/tmp/ansible-tmp-1619617736.3666978-67585791963378/ /var/tmp/ansible-tmp-1619617736.3666978-67585791963378/AnsiballZ_file.py && sleep 0'
[15:48:57] <localhost> EXEC /bin/sh -c 'chown user /var/tmp/ansible-tmp-1619617736.3666978-67585791963378/ /var/tmp/ansible-tmp-1619617736.3666978-67585791963378/AnsiballZ_file.py && sleep 0'
[15:48:57] <localhost> EXEC /bin/sh -c 'sudo -H -S -n -u user /bin/sh -c '"'"'echo BECOME-SUCCESS-ybktoydyqnxllqfblfzirlejobfeqqko; /usr/bin/python /var/tmp/ansible-tmp-1619617736.3666978-67585791963378/AnsiballZ_file.py'"'"' && sleep 0'
[15:49:00] <localhost> EXEC /bin/sh -c 'rm -f -r /var/tmp/ansible-tmp-1619617736.3666978-67585791963378/ > /dev/null 2>&1 && sleep 0'
[15:49:00] STDERR:  [WARNING]: Consider using the unarchive module rather than running tar.  If
[15:49:00] STDERR: you need to use command because unarchive is insufficient you can add
[15:49:00] STDERR: warn=False to this command task or set command_warnings=False in ansible.cfg to
[15:49:00] STDERR: get rid of this message.
[15:49:00] sudo sync
[15:49:00] 
--- Exception Trace ---
Traceback (most recent call last):
  File "run_installation.py", line 377, in run_installation
  File "backend/ansiblecube.py", line 227, in run_phase_two
  File "backend/ansiblecube.py", line 64, in run
  File "backend/qemu.py", line 427, in exec_cmd
backend.qemu.QemuException: ssh command failed with status -1. cmd: sudo sh -c 'cd /var/lib/ansible/local && /usr/bin/ansible-playbook -vvv --inventory hosts --tags move-content,seal --extra-vars="@/var/lib/ansible/local/extra_vars.json" main.yml'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "run_installation.py", line 377, in run_installation
  File "backend/qemu.py", line 169, in __exit__
  File "backend/qemu.py", line 329, in _shutdown
  File "backend/qemu.py", line 413, in exec_cmd
  File "paramiko/client.py", line 493, in exec_command
  File "paramiko/transport.py", line 802, in open_session
  File "paramiko/transport.py", line 896, in open_channel
paramiko.ssh_exception.SSHException: SSH session not active

---
[15:49:00] Restoring system sleep policy
[15:49:00] Resuming xdg-screensaver (wid #0x6400000)
[15:49:00] --> Failed: SSH session not active
[15:49:00] Installation failed: SSH session not active
[15:49:00] [STAGE 5/5: Post-process contents (virtualized) - 100%]
[15:49:00] Call: ['/usr/bin/xdg-screensaver', 'resume', '0x6400000']
[15:49:00] xdg-screensaver: Window 0x6400000 does not exist
[15:49:00] --> Restoring loop device (/dev/loop14) mode
[15:49:00] Call: ['pkexec', 'chmod', '-v', '0660', '/dev/loop14']
[15:49:00] Error executing command as another user: Request dismissed
```

rgaudin, 2021-04-28:
Thanks it looks like the problem we see in github actions: ssh connection between host and guest appear inactive and is then dropped. Will investigate.

rgaudin, 2021-04-28:
Next time please include the full log file please.

rgaudin, 2021-05-07:
Would you mind testing [this build](http://tmp.kiwix.org/ci/kiwix-hotspot-linux-584.tar.gz) ?

rgaudin, 2021-05-07:
@kelson42 confirmed the bug is not systematic: failed twice in a row (early in the process) but the (two?) next attempts succeeded.

Build including 9120fc3 succeeded as well but we can't conclude it fixed it.

As this is sporadic, I am closing this awaiting another report (from @kelson42  or someone else).

One hunch that we had is that it could be linked to the computer going to sleep as the two failed attempts were when the user was not present.

commit 206dc4f6 (2021-04-29): Fixed #584: using bare stdio serial as before
closed 2021-05-07 by rgaudin