offspot/kiwix-hotspot #585: Fails to build with `thorough mount procedure failed`

Popolechien, 2021-04-28:
See https://cardshop.hotspot.kiwix.org/orders/L531R6089
```
[14:52:10] Traceback (most recent call last):
  File "backend/mount.py", line 248, in test_mount_procedure
  File "backend/mount.py", line 452, in mount_data_partition
  File "backend/util.py", line 104, in subprocess_pretty_check_call
  File "backend/util.py", line 97, in subprocess_pretty_call
backend.util.CheckCallException: Process ['/bin/mount', '-t', 'exfat', '/dev/loop1', '/tmp/tmp3gys9l57'] failed

[14:52:10] Installation failed: thorough mount procedure failed
```

Same thing with https://cardshop.hotspot.kiwix.org/orders/L532R6089

rgaudin, 2021-04-28:
Thank you ; the problem is that we mount on the host, calling `/bin/mount` which is version `2.33` on buster (our worker) but PyInstaller included a `libmount.so` inside the self-extracting binary (version `2.31` from ubuntu-18.04 which we use in the CI to build) and that got precedence for ldd.

rgaudin, 2021-04-28:
I've used a workaround for the cardshop (ubuntu-18.04 based worker) so you should be clear of this on the cardshop.
Will require additional work to fix properly though.

rgaudin, 2021-05-03:
```
[14:54:13] [STAGE 1/5: Retrieve Master Image - 10%]
[14:54:13] --> Extracting base image from ZIP file
Extraction complete: /data/608976a998c8bf00234abe34.BUILDING.img
[14:54:42] [STAGE 1/5: Retrieve Master Image - 18%]
[14:54:42] --> Testing mount procedure
. mounting /data/608976a998c8bf00234abe34.BUILDING.img
Call: ['/tmp/_MEIy4KakC/qemu-img', 'info', '-f', 'raw', '/data/608976a998c8bf00234abe34.BUILDING.img']
image: /data/608976a998c8bf00234abe34.BUILDING.imgfile format: rawvirtual size: 8 GiB (8589934592 bytes)disk size: 8 GiBCall: ['/sbin/losetup', '--offset', '7000031744', '--sizelimit', '1585740288', '--find', '--show', '/data/608976a998c8bf00234abe34.BUILDING.img']
/dev/loop1Call: ['/bin/mount', '-t', 'exfat', '/dev/loop1', '/tmp/tmpyjgie5ll']
/bin/mount: /tmp/_MEIy4KakC/libmount.so.1: version `MOUNT_2.33' not found (required by /bin/mount)Call: ['/bin/umount', '/dev/loop1']
/bin/umount: /tmp/_MEIy4KakC/libmount.so.1: version `MOUNT_2.33' not found (required by /bin/umount)Call: ['/sbin/losetup', '--detach', '/dev/loop1']
[14:54:47] Process ['/bin/mount', '-t', 'exfat', '/dev/loop1', '/tmp/tmpyjgie5ll'] failed
[14:54:47] Traceback (most recent call last):
  File "backend/mount.py", line 248, in test_mount_procedure
  File "backend/mount.py", line 452, in mount_data_partition
  File "backend/util.py", line 104, in subprocess_pretty_check_call
  File "backend/util.py", line 97, in subprocess_pretty_call
backend.util.CheckCallException: Process ['/bin/mount', '-t', 'exfat', '/dev/loop1', '/tmp/tmpyjgie5ll'] failed

[14:54:47] Installation failed: thorough mount procedure failed

--- Exception Trace ---
Traceback (most recent call last):
  File "run_installation.py", line 203, in run_installation
ValueError: thorough mount procedure failed
```

commit 7caa83a3 (2021-05-04): Fixed #585: fixed libmount conflict on linux

pyinstaller binary includes a libmount.so.1 as a dependency to some of the
glib shared libraries that gets included with pygobjects.

this file conflicts with the host's libmount that the host's `mount` utility requires
to mount our data parttition. Unless the running host is the same as the building version
(ubuntu-18.04 as of this writting), this results in `mount` not working.

the binary is self-extracted on start then our script is launched with updated
environ so that our app can find included libraries. we have no control over this.

This co [... cut]
commit ef1bfdb8 (2021-05-04): Fixed #585: fixed libmount conflict on linux

pyinstaller binary includes a libmount.so.1 as a dependency to some of the
glib shared libraries that gets included with pygobjects.

this file conflicts with the host's libmount that the host's `mount` utility requires
to mount our data parttition. Unless the running host is the same as the building version
(ubuntu-18.04 as of this writting), this results in `mount` not working.

the binary is self-extracted on start then our script is launched with updated
environ so that our app can find included libraries. we have no control over this.

This co [... cut]
closed 2021-05-04 by rgaudin