offspot/kiwix-hotspot #590: Master fails to build in CI

rgaudin, 2021-05-03:
For some reason, master creation fails to run in Github Actions.

https://github.com/kiwix/kiwix-hotspot/runs/2491520908?check_suite_focus=true

The log suggests that the SSH connection between hotspot and QEMU gets lost during a long task.

```
TASK [system : Install or upgrade all required package] ************************
task path: /var/lib/ansible/local/roles/system/tasks/main.yml:65
<localhost> ESTABLISH LOCAL CONNECTION FOR USER: root
<localhost> EXEC /bin/sh -c 'echo ~root && sleep 0'
<localhost> EXEC /bin/sh -c '( umask 77 && mkdir -p "` echo /root/.ansible/tmp/ansible-tmp-1620041282.44715-244723419476936 `" && echo ansible-tmp-1620041282.44715-244723419476936="` echo /root/.ansible/tmp/ansible-tmp-1620041282.44715-244723419476936 `" ) && sleep 0'
Using module file /usr/lib/python3/dist-packages/ansible/modules/packaging/os/apt.py
<localhost> PUT /root/.ansible/tmp/ansible-local-2674543ouv7m/tmpa3p22mpq TO /root/.ansible/tmp/ansible-tmp-1620041282.44715-244723419476936/AnsiballZ_apt.py
<localhost> EXEC /bin/sh -c 'chmod u+x /root/.ansible/tmp/ansible-tmp-1620041282.44715-244723419476936/ /root/.ansible/tmp/ansible-tmp-1620041282.44715-244723419476936/AnsiballZ_apt.py && sleep 0'
<localhost> EXEC /bin/sh -c '/usr/bin/python /root/.ansible/tmp/ansible-tmp-1620041282.44715-244723419476936/AnsiballZ_apt.py && sleep 0'
sudo sync
/tmp/_MEIFxPNKt/paramiko/kex_ecdh_nist.py:39: CryptographyDeprecationWarning: encode_point has been deprecated on EllipticCurvePublicNumbers and will be removed in a future version. Please use EllipticCurvePublicKey.public_bytes to obtain both compressed and uncompressed point encoding.
/tmp/_MEIFxPNKt/paramiko/kex_ecdh_nist.py:96: CryptographyDeprecationWarning: Support for unsafe construction of public numbers from encoded data will be removed in a future version. Please use EllipticCurvePublicKey.from_encoded_point
/tmp/_MEIFxPNKt/paramiko/kex_ecdh_nist.py:111: CryptographyDeprecationWarning: encode_point has been deprecated on EllipticCurvePublicNumbers and will be removed in a future version. Please use EllipticCurvePublicKey.public_bytes to obtain both compressed and uncompressed point encoding.
[11:28:40] ERROR: unable to properly create image: SSH session not active
Error: Process completed with exit code 1.
```

Only the first two and last two lines are of interest. It started an apt task which installs a lot of packages.
Works fine of other environments but GHA as far as we've seen.

kelson42, 2021-05-08:
@rgaudin Might that be that last fix of #584, fixes this as well?

kelson42, 2021-07-26:
This ticket will be invalidate if we move to #429 Docker based solution. Wonder if we should do anything anymore for this ticket.

kelson42, 2021-12-03:
In 2.6, we plan to build the master using pygen. Should be a lot quicker/robuster. Hopefuly the CI won't die anymore at this stage, using this approach.

closed 2021-12-03 by kelson42