offspot/kiwix-hotspot #594: Replace etcher-cli

kelson42, 2021-05-05:
Etcher-cli is not maintained anymore

kelson42, 2021-05-05:
@rgaudin https://github.com/raspberrypi/imagewriter seems to be a good candidate, but can it write a randome image? Should we request via a ticket a non-interactive mode?

rgaudin, 2021-05-05:
It can write any image but has no CLI. Requesting one wouldn't hurt obviously.

kelson42, 2021-05-05:
Here it is https://github.com/raspberrypi/rpi-imager/issues/221

kelson42, 2021-05-05:
@rgaudin I believe it might be helpful to provide a mimal option set on the upstream ticket. We should not mockup all options available in the GUI, but only the minimal which is probably choose the image and the output device.

kelson42, 2021-05-06:
We can replace etcher-cli by balena-cli, but even if maintained, this one is not notarised either. There is ticket to request it https://github.com/balena-io/balena-cli/issues/2260.

rgaudin, 2021-05-07:
https://github.com/raspberrypi/rpi-imager/issues/221 has been implemented and https://github.com/balena-io/balena-cli/issues/2260 was denied.

We should thus consider going with rpi-imager which now have a cli option that works.
What would be required:

- adapt code to new cli
- adapt CI to bundle it
- [macos] support sudo (currently requires GUI interaction)
- [macos] bundle full rpi-imager.app
- [linux] compile static version (`libarchive13 (>= 3.0.4), libc6 (>= 2.14), libcurl3-gnutls (>= 7.16.2), libgcc-s1 (>= 3.0), libqt5core5a (>= 5.12.2), libqt5dbus5 (>= 5.0.2), libqt5gui5 (>= 5.2.0) | libqt5gui5-gles (>= 5.2.0), libqt5network5 (>= 5.0.2), libqt5qml5 (>= 5.1.0), libqt5widgets5 (>= 5.0.2), libssl1.1 (>= 1.1.0), libstdc++6 (>= 5.2), qml-module-qtquick2, qml-module-qtquick-controls2, qml-module-qtquick-layouts, qml-module-qtquick-templates2, qml-module-qtquick-window2, qml-module-qtgraphicaleffects, dosfstools, fdisk, udisks2`)

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42