offspot/kiwix-hotspot #595: Order fails with installer rc-1 notification

Popolechien, 2021-05-05:
See https://cardshop.hotspot.kiwix.org/orders/L545R6092

```
Starting to work on 609298e898c8bf23184a986d
updating task #609298e898c8bf23184a986d status to: building
Starting image creation
Starting /usr/bin/kiwix-hotspot cli --root --filename 609298e798c8bf23184a9847 --build-dir /data --config /data/609298e798c8bf23184a9847.json --size 256GB

installer started: <subprocess.Popen object at 0x7f03dcf2bb80>
installer failed: 1
collecting full terminated log
installer rc: 1
Traceback (most recent call last):
  File "worker/tasks/create.py", line 108, in run
    getattr(self, method)()
  File "worker/tasks/create.py", line 200, in build_image
    raise subprocess.SubprocessError("installer rc: {}".format(ps.returncode))
subprocess.SubprocessError: installer rc: 1
updating task #609298e898c8bf23184a986d status to: failed_to_build
job is not alive
job crashed: installer rc: 1
```

rgaudin, 2021-05-05:
It's a worker error: `Not enough space available at /data (457.03 GiB) to build image (475.75 GiB)`

rgaudin, 2021-05-05:
Reclaimed some space on the worker ; free space now at 490GB. Tight be should work. Please recreate the order.

Not a hotspot issue.

closed 2021-05-05 by rgaudin