offspot/kiwix-hotspot #598: Zimit-generated files cannot be read/accessed on hotspot

Popolechien, 2021-05-07:
I generated an image [L548R6094](https://cardshop.hotspot.kiwix.org/orders/L548R6094) containing the zimit-generated GCFglobal zim: the [zim works perfectly on library.kiwix.org](https://library.kiwix.org/edu.gcfglobal.org_en_all_2021-03) but returns 
[image]

when trying to access from the hotspot

rgaudin, 2021-05-07:
https://github.com/openzim/warc2zim/issues/24

kelson42, 2021-05-07:
@rgaudin Yes, but we have no ticket to have HTTPS running on the kiwix-hotspot right?

rgaudin, 2021-05-07:
Hotspot doesn't support SW zims. @kelson42 should we exclude `_sw:yes` in the Cardshop for now?

rgaudin, 2021-05-07:
> @rgaudin Yes, but we have no ticket to have HTTPS running on the kiwix-hotspot right?

Couldn't find any, so I'd say no.

kelson42, 2021-05-07:
We need to listen on https as well witth a self signed cert.

rgaudin, 2021-05-07:
> We need to listen on https as well witth a self signed cert.

We already have this. We just need to enable SSL for all vhosts

rgaudin, 2021-05-07:
And maybe add a wildcard to the cert so that there's only the self-CA that triggers an issue

PR #599 (2021-05-07): Added HTTPS support for all vhosts
All vhost now reachable via HTTPS URLs, using the single, shared certicicate
that we used for the captive portal.

Also updated the list of altnames in the certificate to include all of those but that's
not strictly necessary as our self-signed certificate is already refused by browsers
for that reason and the manual bypass is general.

Fixes #598
commit 3052d474 (2021-05-07): Fixed #598: added HTTPS support for all vhosts

All vhost now reachable via HTTPS URLs, using the single, shared certicicate
that we used for the captive portal.

Also updated the list of altnames in the certificate to include all of those but that's
not strictly necessary as our self-signed certificate is already refused by browsers
for that reason and the manual bypass is general.
closed 2021-05-07 by rgaudin