offspot/kiwix-hotspot #602: Can't install Kiwix hotspot on Sd card

LuX8712, 2021-05-13:
i have a 16gb sd card, and i want to install around 12 gb of data on it, with around 2.4 left. I try to run the software, but i get this error 

 Invalid SD card size: 14.4 GiB
SD card size has to be a power of 2, e.g. 16 GiB.
You can resize disk images with 'qemu-img resize <imagefile> <new-size>'

please help. idk what to do

rgaudin, 2021-05-13:
Which version are you using ? It's in the File / about

LuX8712, 2021-05-13:
I'm using the latest version of the software, 2.0.

Popolechien, 2021-05-13:
Latest version is 2.4.3 IIRC
https://download.kiwix.org/release/kiwix-hotspot/

rgaudin, 2021-05-13:
@LuX8712 I suppose you meant `2.4.0`. Please use the 2.4.3+ version which includes a fix for this.

LuX8712, 2021-05-13:
this did not work.

rgaudin, 2021-05-13:
Would you mind being a little more verbose? Do you mean you installed version 2.4.3 and got the same error ?
Could you share a screenshot of the About window displaying the version and paste the log here ? There's a button to copy the log to the pasteboard.

Thank you

LuX8712, 2021-05-13:
Yes, i installed 2.4.3 and the same error occurred. I will  reattempt this to see if i get the error code again, and post it here.

LuX8712, 2021-05-13:
[LOG.txt](https://github.com/kiwix/kiwix-hotspot/files/6471249/LOG.txt)

That is the log i copied.

[image]

the version

rgaudin, 2021-05-13:
Thank you ; will come back to you

kelson42, 2021-05-14:
@LuX8712 we believe to understand the problem and therefore confirm there is a bug. Fix to come soon.

commit b28ac758 (2021-05-14): Fixed #602: working with an SD card (not a file) using GUI
closed 2021-05-14 by rgaudin