offspot/kiwix-hotspot #607: RPI does not boot after succesfull run

kelson42, 2021-05-20:
With:
* Kiwix Hotspot 2.4.5
* Ubuntu 20.04
* A new 16GB SD card (which has never ever been tested earlier with Kiwix Hotspot)

Here is what I get:
[image]

Log is:
[broken-boot.log](https://github.com/kiwix/kiwix-hotspot/files/6513294/broken-boot.log)

kelson42, 2021-05-20:
Article about this behaviour https://samx18.io/blog/2017/11/05/piBootIssue.html

Di9italZero, 2021-05-22:
I am having the same issue. After downloading the hotspot builder and selecting the options I get this same exact error after booting from a pi 4 4gb. Seems the start elf issue has now been replaced with this. It's been 6 months I've tried to get this to work on a pi 4 using the hotspot, not kiwix serve or kiwix tools. Is this going to be fixed or should i just move on?

rgaudin, 2021-05-22:
@Di9italZero are you also using 2.4.5 GUI version on Ubuntu 20.04?

Di9italZero, 2021-05-22:
No. I am using windows to create the Hotspot. I just left the settings as they were, I selected my usb flash drive. I selected the desktop as the working path. I only selected one Ted Talk as last time I attempted this it took hours to complete and then failed. The program completed successfully. I then removed the USB from my windows machine and placed it into my the USB port on my raspberry pi 4 4g. It is configured to boot from USB first. When applying power the start elf error has been resolved however now it gives the same exact error as listed above. Its states it unable to access the system and asks me to press enter where the process repeats.

kelson42, 2021-05-23:
We will need to fix it, but this might take a good month before the next release of 2.4.6. What would be good for the moment is:
* Know if this is a systematic bug
* Do we have a workaround? Does the manual writting of the image fixes the problem?

Di9italZero, 2021-07-22:
Any progress yet?

Di9italZero, 2021-08-16:
Is this project abandoned?

kelson42, 2021-08-16:
@Di9italZero Like you can see this ticket is part of the next milestone. If you want to get this fixed even quicker, please make a PR.

closed 2024-06-04 by kelson42