offspot/kiwix-hotspot #608: Hotspot does not read zimit-generated files

Popolechien, 2021-05-26:
Created an image using installer that had the [GCFD zim](https://library.kiwix.org/edu.gcfglobal.org_en_all_2021-03/A/index.html) (zimfarm receipe [here](https://farm.openzim.org/recipes/edu.gcfglobal.org_en)) 

Installation ran fine and upon opening the zim I get the expected "switch to https" warning. But then past this step I only get the little circle going round and round indefinitely and there is no access to content.

rgaudin, 2021-05-26:
Doesn't look like a hotspot issue but downloading the ZIM to make sure and maybe transfer.

rgaudin, 2021-06-02:
Can't reproduce ; would you mind retesting but opening the developer tools console first (⌥ + ⌘ + I  on Firefox) and copy/screenshot its output ? So we can see where it's stuck at.

Popolechien, 2021-06-02:
It actually works on Firefox, but not on Chrome.
Looking at the dev console I get this:
`load.js:56 Uncaught (in promise) DOMException: Failed to register a ServiceWorker for scope ('https://kiwix.hello.hotspot/edu.gcfglobal.org_en_all.en/A/') with script ('https://kiwix.hello.hotspot/edu.gcfglobal.org_en_all.en/A/sw.js?replayPrefix=&root=edugcfglobal.org_en_all.en'): An SSL certificate error occurred when fetching the script.`

kelson42, 2021-07-25:
@rgaudin Does this bug is confirmed? I would be really surprised!

rgaudin, 2021-07-27:
Is the symptom the same for all of them ?

Popolechien, 2021-07-27:
Yup. Little round thing going to no end.

rgaudin, 2021-07-27:
Ok, you tried different browsers ?

Popolechien, 2021-07-27:
Ok, got it:

Chrome (macOS, all extensions deactivated): does not work
Firefox (macOS), Chrome (android): I get the below activation page

[image]

This being said, Chrome is the dominant browser so it might be worth checking if it is an issue with macOS or if it happens on Windows as well.

rgaudin, 2021-07-27:
Thanks ; that's helpful. Pretty sure it happens on all platform ; your initial message make it seems like it.

Popolechien, 2021-07-27:
Yep, can confirm that Opera on PC (windows 10) is fine, but not Chrome (Win10).

kelson42, 2021-07-28:
@rgaudin Both of us have been quite weak on that one :( @Popolechien Sorry for this pretty stupid and high impacting problem.

rgaudin, 2021-09-30:
> Chrome (macOS, all extensions deactivated): does not work

Works fine for me, and that does not surprise me as I did test it when implementing the feature (had to take a screencast of it!)

[image]

Are you sure you had the problem on macOS? Trying to get a Windows running to test on Windows now.

rgaudin, 2021-09-30:
OK, so the problem is after this tutorial, which works on all browsers. It's just that once you've added the exception, you don't see it anymore obviously.

I can reproduce the issue: certificate exception is added but chrome refuses to register the service worker still.

rgaudin, 2021-09-30:
OK, unfortunately, Chrome now requires the certificate to be installed on the system to work with Service Workers.
The process is thus more complicated with many steps and of course it's different on Windows and macOS.

What should we do?

Popolechien, 2021-09-30:
well the simpler for the user the better, no matter the cost on our side.

rgaudin, 2021-09-30:
There is no simple way for the user.

The current (other browsers though) solution requires a few clicks in the browser to add an exception for the domain.
What we are looking at now is a many steps, multi-software process to get to the same result:

1. Click on “Not secure” button next to address bar
1. Click on “Certificate (invalid)”
1. In the popup, go to Details
1. Click export to file
1. Select a location and name to export to and save
1. Now in explorer, go to that location and double-click certificate
1. Choose install certificate
1. Select “place certificate in following store”
1. Click Browse
1. Select “Trusted root certificates authorities” from the list of similarly named options
1. Confirm
1. Confirm in the warning dialog
1. Exit the certificate manager
1. In chrome, refresh the page you were on

Steps 7-13 are visible [in this tutorial](https://techcommunity.microsoft.com/t5/windows-server-essentials-and/installing-a-self-signed-certificate-as-a-trusted-root-ca-in/ba-p/396105)

Another alternative, is to ask user:
- To use another browser (Firefox)
- To disable security in chrome by launching it with `--ignore-certificate-errors`.

kelson42, 2024-06-04:
Invalidated with Zimit2