offspot/kiwix-hotspot #610: Something is wrong with configurations size calculation

Popolechien, 2021-05-28:
I am looking at https://cardshop.hotspot.kiwix.org/configurations/327 which has the following contents:
```
Nomad android app (50 MiB) 
Math Mathews android (50 MiB) 
Africatik apps (2.7 GiB) 
KA Lite fr (11.35 GiB)
cest-pas-sorcier_fr_all.fr (74.7 GiB)
jaimelire_fr.fr (70.49 MiB)
japprendsalire_fr.fr (109.45 MiB)
les-fondamentaux_fr_all.fr (126.63 MiB)
vikidia_fr_all_maxi.fr (801.7 MiB)
wikipedia_fr_all_maxi.fr (28.91 GiB)
education-et-numerique_fr_all.fr (3.52 giB)
wiktionary_fr_app_nopic.fr (1.13 GiB)
```

If I had these file sizes, I come to a rough total of 124 Gib (make it 132 GiB when adding the 8GiB default image setup). The card size is however computed to take 147,88 GiB.

If I poke around and try to build the same config from scratch, it looks like the addition of Khan Academy is the one that causes the largest discrepancy.

rgaudin, 2021-05-28:
Replicated your config and got this
[image]

What is your concern exactly ? That the sum of all content is not equal to the total size indicated?
Please share a config.json file so that I can give you exact number about each content and what margins we add and the reasons for those.

Popolechien, 2021-05-28:
JSON is not supported here, will send via Slack

rgaudin, 2021-05-28:
```json
{
    "name": "Something",
    "project_name": "Some name",
    "language": "fr",
    "timezone": "Europe/Paris",
    "wifi_password": null,
    "admin_account": {
        "login": "abcdefgh",
        "password": "abcdefgh"
    },
    "size": "200 GB",
    "content": {
        "zims": [
            "cest-pas-sorcier_fr_all.fr",
            "jaimelire_fr.fr",
            "japprendsalire_fr.fr",
            "les-fondamentaux_fr_all.fr",
            "vikidia_fr_all_maxi.fr",
            "wikipedia_fr_all_maxi.fr",
            "education-et-numerique_fr_all.fr",
            "wiktionary_fr_app_nopic.fr"
        ],
        "kalite": [
            "fr"
        ],
        "wikifundi": [],
        "aflatoun": true,
        "edupi": false,
        "edupi_resources": null,
        "nomad": true,
        "mathews": true,
        "africatik": true
    },
    "branding": {
        "logo": null,
        "favicon": null,
        "css": null
    }
}
```

kelson42, 2024-06-04:
This bug has been fixed in image-creator

closed 2024-06-04 by kelson42