offspot/kiwix-hotspot #611: Cant create image for Rasberry Pi

jannis72929, 2021-06-01:
So I have the problem that the installer wont create the image. After a few attempts I always get the error code: 
_--> Failed: 'charmap' codec can't decode byte 0x81 in position
 Installation failed: 'charmap' codec can't decode byte 0x81 in position 290: character maps to <undefined>_

rgaudin, 2021-06-01:
@jannis72929 thank you for your report. What you shared is not enough to reply. Could you share the complete log? There's a button in the installer to copy it or export it to a file.

jannis72929, 2021-06-01:
[log.txt](https://github.com/kiwix/kiwix-hotspot/files/6577721/log.txt)

rgaudin, 2021-06-02:
Thank you; it seems that the output of imdisk's formatting include a char that cannot be represented in cp1252…
What doesn't feel right is that the output should be something like:
```
[18:05:25] Creating device...
[18:05:25] Created device 0: F: -> D:\\hotspot-2021_06_01-17_18_58.BUILDING.img
[18:05:25] Notifying applications...
[18:05:25] Done.
```
which decodes fine. Maybe it's different on a German Windows ; will have to check.

kSunrice, 2021-07-17:
hello,
i had the same problem but i think i found a workaround.
my windows 10 normally is german and on windows 7, also german, i got the same problem.
i switched the systemlanguage to english, reboot the system and tried it again.
now the image was succesful created.. 
hope it helps ;)

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42