offspot/kiwix-hotspot #612: Logfile/tracking Wikipedia usage

atharijeff, 2021-06-21:
The Literate Earth Project operates a network of 16 primary school based libraries in Uganda. We are rolling out Raspberry Pi with Kiwix, mainly to give students access to an offline version of Wikipedia.

We a plethura of funders who want us to expand our work at other schools but we have no metrics, other than sign-up sheets for using the Raspberry Pi units, that students are utilizing it. We need some way to track searches and when a student clicks a hyperlink. Perhaps this could just be something that tracks each time the URL changes.

We have spoken with more seven organizations that also use this model in Uganda. They have said they would benefit from tracking. I have spoken with the Raspberry Pi's International Programme Manager, James Aslett. He also is not aware of a way to track this and notes the value for funders for organizations using Kiwix all over the world.

rgaudin, 2021-06-21:
That's something we want to do through #429. cc @Popolechien

atharijeff, 2021-06-21:
Fantastic. Glad there is other movement for this @rgaudin. Please let me know if you need me to delete this.

Popolechien, 2021-06-29:
See #510 as well.

atharijeff, 2021-07-09:
Per Stephane's guidance, elaborating here:

We ultimately need to track URL changes. That will allow us to understand how many searches or hyperlink clicks (which is basically a search/pulling up new info). It needs to be stored on a file that our staff can extract every three months when they visit the sites.

kelson42, 2021-08-03:
Seems a duplicate of #510

closed 2021-08-03 by kelson42