offspot/kiwix-hotspot #615: Allow image creation that includes locally-stored file

Popolechien, 2021-07-15:
Should a user generate a zim file with zimit, they should be able to include said file (presumably locally stored) to their configuration and have it in the end image.

kelson42, 2021-07-15:
After having discussed this with @rgaudin and @mgautierfr it seems that we should allow to do that in the CMS, and the the cardshop will just follow.

rgaudin, 2021-07-15:
@kelson42 this is not the Cardshop, this is kiwix-hotspot. Users manually firing the hotspot tool should IMO have a relatively easy way to add non-published zim files. A bit different than cardshop use case.

kelson42, 2021-07-15:
@rgaudin Agree

kelson42, 2024-06-04:
Duplcate of #664 

Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42