offspot/kiwix-hotspot #616: New upper limit for devices

Popolechien, 2021-07-22:
user reported that they could not connect more than eight devices simultaneously on a RPi 3B+. Are we aware of any upgrade that could explain this (and could there be a fix)?

rgaudin, 2021-07-22:
Linking #544 and https://github.com/iiab/iiab/issues/823

rgaudin, 2021-07-22:
@Popolechien could you get confirmation that it cannot **connect** the ninth devices to WiFi ? What does the Device says ? Does it refuses the connection ? Is it getting the WiFi connection but no IP, do it stales in connecting, etc.
It's important we know __exactly__ where that limit is.

We know there is the above-mentioned issue that seems to be driver/firmware originated but it can certainly be something else. According to that ticket, the limit should be 10, not 8.

Popolechien, 2021-07-22:
@rgaudin Tablet does not see the network and does not list it.

rgaudin, 2021-07-22:
Thanks. Looks like we should try IIAB's fix (reverting to an older firmware)

Popolechien, 2021-07-22:
tested with an older device as 9th device and it picked up the signal. so yeah, probably an issue with the newer firmware.

commit 9186129f (2021-07-30): Fixed #616: using older firmware for broadcom devices

As per ticket discussion, buster provided firmware doesn't allow more than 10
simultaneous connexions to the WiFi AP.

This replaces the firmware with an older version that doesn't have this limit and should
allow something like 32 WiFi connexions.

Testing which firmware is loaded via `ethtool -i wlan0`

New, limited firmare:
driver: brcmfmac
version: 7.45.229
firmware-version: 01-2dbd9d2e

Older, non-limited one
driver: brcmfmac
version: 7.45.18
firmware-version: 01-6a2c8ad4
closed 2021-07-30 by rgaudin