offspot/kiwix-hotspot #617: Edupi admin access does not meet user manual's instruction

Popolechien, 2021-07-25:
Edupi's user manual indicates that it is possible to access the software's backend by going to edupi.fondationorange.org/custom and using pi / raspberry as login and password (see p.10 of attached doc). According to Orange, this is not working. Has anything changed in that regard?
[Manuel Edupi Fondation Orange mars 2017.pdf](https://github.com/kiwix/kiwix-hotspot/files/6874473/Manuel.Edupi.Fondation.Orange.mars.2017.pdf)

rgaudin, 2021-07-25:
- Edupi is not our software
- We didn't write this document (and it's dated 2017).
- edupi.**fondationorange.org** doesn't belong to us.

If you want to access the *admin* of an EduPi installed on a kiwix-hostpot, yes, `edupi.<name>.hotspot/custom` is the right address and should work.

Popolechien, 2021-07-25:
Good enough, thanks.

closed 2021-07-25 by Popolechien