offspot/kiwix-hotspot #618: Remove network name from landing page

Popolechien, 2021-08-19:
The default configuration looks like Kiwix / kiwix - looks awkward and does not bring any relevant information

[image]

rgaudin, 2021-08-19:
I think it's probably the most customized option so it's more frequent than not that it is not `Kiwix`.

I believe it's useful as this matches the WiFi network and the URL to type to get in so it probably helps to have it on screen.

If what annoys you is the fact that we have Kiwix twice, maybe the problem is with the UI (position, logo) and not with the information itself?

kelson42, 2021-08-19:
@rgaudin I think there is indeed a "problem" and this is not the first time we talk about that. What is exactly the meaning of this "KIWIX" string on the right? Does that is explained somewhere for the end-user?

rgaudin, 2021-08-19:
That's the name of the hotspot. It used to say `{name}.hotspot` but the tld was removed upon request.
`kiwix` is the default name on the cardshop. The name is used as SSID as well. It is customized most of the time (I believe, haven't checked numbers).

The Kiwix logo is not customized often though IMO.

This UI is inherited (copied) from ideascube's one. I see some value in having the name displayed on home page:
- Marks it in user's mind. Important information to recall where to go when you've left the homepage. It's not written anywhere else (except URL). Might be useful as we've increased captive-portal timeout.
- In case there are multiple Hotspots reachable, indicates which one you're on.

I'm not opposed to remove it at all, but « double-kiwix » isn't much convincing. Why not remove the Kiwix logo instead?

kelson42, 2021-08-19:
We have to few custimisation ability. Therefore I'm against remove the logo. We would have to keep it anyway only because/for FO.

I'm in favour of removing the name of the hotspot. If end-user see this, he already knows about the SSID. Carshop name is not relevant here fore end-user.

rgaudin, 2021-08-19:
> We have to few custimisation ability. Therefore I'm against remove the logo (I would say, hide it in CSS per default). We would have to keep it anyway only because/for FO.

The problem is that we have kiwix twice and now you're talking about Fo logo… We used to have a logo only when customized and it was requested here that we add Kiwix one if there is no customization.

This really feels like we're playing rubikcube without knowing what we're doing

> I'm in favour of removing the name of the hotspot. If end-user see this, he already knows about the SSID. Carshop name is not relevant here fore end-user.

Don't know why you're mentioning the cardshop here. Assumption that end-user knows about the SSID is far-fetched but we've reached consensus on removing this as “non-valuable information”. Works for me.

Popolechien, 2021-08-19:
My experience is that most "corporate" (as in, using the hotspot for an org rather than their own purpose) do use the logo, which more often than not bears the name of the network (case in point: Fondation Orange, which currently has 2-300+ hotspots that go FO/FO).

I understand the point about the useful info (eventhough I would disagree), but right now the ticket is more about UI. If you want to move the SSID to the top or bottom right, fine by me.  I would merely suggest being more explicit and display it clearly as `http://_name_.hotspot` or  `name Network`

rgaudin, 2021-08-19:
OK then let's remove it as you suggested.

kelson42, 2021-08-19:
Yes, hide it in CSS and if someone really wants it, then easy to reintroduce. If at somepoint we want to reintroduce it somewhere else in the UI layout, this will be easy as well.

commit f58accba (2021-11-25): Fixed #618: Removed (hide) project name in home page
closed 2021-11-25 by rgaudin