offspot/kiwix-hotspot #624: Cache directory should be configurable

kelson42, 2021-11-03:
Currently, it is not, the cache directory is in the (image) output directory. This brings problems:
* This is not configurable and it exists various scenarios where this is not wanted
* It is "hidden", ie. if the user does not look accurately to the fs, it can miss the fact that there is a cache (just had a user report about that).
* The way the installer behaves is implicit

IMO the current behaviour is an appropriate default behaviour. But it should be visible/configuratble in the UI/cmd. I would propose to put it below the output directory file picker. Once the output directory has been set, configure the cache directory automatically (if not already configured manually). 

This ticket is a follow of #623.

rgaudin, 2021-11-03:
I find a single user report not enough to trigger a UI change, especially given our window is already packed and hard to use on many screens.

I'd postpone that to the version with the UI makeover.

kelson42, 2021-11-03:
Yes, definitly not urgent. But the way it works now is not explicit enough. I had a doubt myself. Next UI shoudl probably have a simple and an advanced view or implement any other approach to not display all the options.