offspot/kiwix-hotspot #627: Integrate Datapost (or similar data collection tool)

Popolechien, 2021-11-30:
World Possible's Rachel has deployed a very neat tool called Datapost that allows people to connect to a box and download a set of metrics on their Android device. From there, they can carry it to the next wi-fi enabled area and upload that data so that the devices can "ping". They [uploaded a short video](https://www.youtube.com/watch?v=COt3UqdE-N8) explaining the basics.
[image]

[image]

I've sent them a note asking for more info (could not find their main dev on here unfortunately)

kelson42, 2021-12-04:
@Popolechien Similar ro #510, no?

Popolechien, 2021-12-04:
@Kelson Ah yes. Should I update #510 listing this specific solution?

kelson42, 2021-12-04:
@Popolechien Not necessary, both tickets will be linked.

closed 2021-12-04 by kelson42