offspot/kiwix-hotspot #628: WikiFundi filenames fails to write on host-mounted data partition

rgaudin, 2021-12-02:
Mediawiki stores filenames using user-provided name. It is very permissive so there is no control over what characters can be used in the filename. As long as the file can be stored on the server's filesystem, it's accepted.

This means that our wikifundi archives (tar.bz2) may contain filenames that are not accepted in every circumstances. We extract those archive, from the host, onto the host-mounted data partition which is an exfat partition.

Exfat itself is unicode and [despite what WP says](https://en.wikipedia.org/wiki/ExFAT), those *forbidden characters* (`?<>\:*/"`) are not exfat-incompatible but Windows-incompatible.

It is thus possible to write such characters on an exfat partition mounted on macOS (default) or on Linux (granted it was mounted as unicode).

So we won't be able to write those files onto the partition in Windows ; and Linux (desktop) is also problematic (see below). We need an alternative.

## Option A: fix mediawiki and recreate archives

This mediawiki bug is long-know https://github.com/offspot/wikifundi/issues/72 and @kelson42 opened an [upstream ticker](
https://phabricator.wikimedia.org/T199140) but the proposed solution doesn't work (tried it when upgrading MW to 1.36.1).

While this might be a long-term solution, it's not possible right-now.

## Option B: drop those file

In wikifundi 2021-11 (accross the three languages), there are 19 files concerned:

```
Pancartas_8M_de_2020_"Soy_la_artista,_no_la_musa"_y_"Sin_mujeres_Wikipedia_no_es_una_enciclopedia".jpg
¿Qué_es_Wikipedia?.ogv
¿Cómo_agregar_referencias_dentro_de_Wikipedia?.webm
Bild_från_familjen_von_Hallwyls_resa_genom_Algeriet_och_Tunisien,_1889-1890.._"Vy_över_Alger."_-_Hallwylska_museet_-_91850.tif
¿Cómo_poner_enlaces_internos_dentro_de_Wikipedia?.webm
Pancartas_8M_de_2020_Soy_la_artista,_no_la_musa_y_Sin_mujeres_Wikipedia_no_es_una_enciclopedia.jpg
Ambox_?.svg
Washington,_D.C.,_April_1865_(i.e._1863?)_LCCN2013651864.tif
Ford_Model_"T"_car_no_2_on_display,_Alaska_Yukon_Pacific_Exposition,_Seattle,_June_1909_(AYP_502).jpeg
Personnages_de_la_scène_de_Janvier_(Frères_Limbourg_?)_-_Très_Riches_Heures_du_duc_de_Berry_(f.1).jpg
Conference_of_the_Big_Three_at_Yalta_makes_final_plans_for_the_defeat_of_Germany._Here_the_"Big_Three"_sit_on_the..._-_NARA_-_531340.tif
Volcán_Chimborazo,_"El_Taita_Chimborazo".jpg
"Coconut"_Chair,_1958.jpg
Mona_Lisa_Restored_Colour._Based_on_newly_discovered_"Prado_Copy"_painted_by_pupil_alongside_Leonardo..jpg
Ford_Model_"T"_car_no._2,_winner_of_the_1909_trans-continental_race_from_New_York_to_Seattle.jpg
Creative_Commons_"Non-Commercial"_icon_(SVG,_nominal_64x64).svg
Rembrandt_-_The_Little_Children_Being_Brought_to_Jesus_("The_100_Guilder_Print")_-_WGA19060.jpg
"The_School_of_Athens"_by_Raffaello_Sanzio_da_Urbino.jpg
"The_patient's_skin_is_burned_in_a_pattern_corresponding_to_the_dark_portions_of_a_kimono_worn_at_the_time_of_the_explos_-_NARA_-_519685.jpg
```

It seems the only culprit characters are `?` and `"`. 

From this list, only the three spanish videos are wikifundi-added content ; other are illustrations from articles. Those 3 videos are actually duplicate and are not used: the `?` caused other issues and were easily spotted and thus renamed in wikifundi.

This option proposes to remove those 19 files (so, loosing 16) from the archives and go on. That's [what we've been doing in the past](https://github.com/offspot/kiwix-hotspot/wiki/support-files#wikifundi-otherwikifundi).

## Option C: fix in wikifundi

We could imagine an extra-step in wikifundi-import that would loop over all files, identify those that have incompatible characters, then rename them. It would also need to iterate over all the articles that include them and make the proper substitutions in their wikitext.

It's a lot of effort (work and tests) to work around a mediawiki bug.

## Option D: fix in hotspot

Another option would be to extract files into safe-paths. While the archive might contain unsafe names, we'd unpack to generated names (digest or uuid or any other hashing) and write a mapping file containing the generated paths with their actual counter-part. At runtime (could be in qemu or on device), if this mapping file is present, a script would rename all those files to the proper ones and remove the mapping file.

This looks like a good long-term solution if MW can't be fixed or if we face this with other non-mediawiki archives.

## Unicode on exfat Linux

Writting this here as reference for archiving purposes.

* when using our *root-mode*, we can fix unicode on exfat by appending the `iocharset=utf16` option to the `mount` command. Works 👍 
* when not using it (desktop mode), we rely on udisks2 for which it should be possible to set option (`udiskctl mount -o iocharset=utf16`) but is actually not possible because the loop-device creation automatically mounts the partition (so no control over the mount options) and the unmount command automatically detaches the loop-device (so not possible to remount manually). Playing with udisks daemon config is probably not what we want (and tests were unsuccessful anyway) neither would be changing UDEV's config to disable automounting. Most realistic option would be to use *root-mode* either by  preventing desktop mode (alert on start if not root) or refactor code so the actual creation is launch in a *root-mode* subprocess from the GUI.

Popolechien, 2021-12-02:
Ok so A is not a short term option, B is not a quality option, C is not a practical option -> let's go for D then? Can this fix be implemented fairly quickly though?

rgaudin, 2021-12-03:
Discussing this with @kelson42, we've decided to go with B. for this round, keeping features in par with previous version ; but we'll do C. for the next import round.

closed 2021-12-03 by rgaudin