offspot/kiwix-hotspot #630: Creator crashes at start

Popolechien, 2021-12-19:
When launching https://cardshop.hotspot.kiwix.org/orders/L763R61BF I immediately got a crash with the following (and up to this day never seen) error:
```'''[19:30:39] --> Prepare Image file
Call: ['/sbin/losetup', '--find']
/dev/loop8[19:30:39] --> Change loop device mode (None)
[19:30:39] Installation failed: [Errno 2] No such file or directory: '/dev/loop8'

--- Exception Trace ---
Traceback (most recent call last):
  File "run_installation.py", line 121, in run_installation
  File "backend/mount.py", line 180, in allow_write_on
FileNotFoundError: [Errno 2] No such file or directory: '/dev/loop8'
````'''

rgaudin, 2021-12-19:
Ah that must have been me, testing my Bard image stuff on the worker host. Please retry.

Popolechien, 2021-12-19:
Nope. Same issue with https://cardshop.hotspot.kiwix.org/orders/L764R61BF

rgaudin, 2021-12-20:
Sorry I didn't clean it correctly but that was the reason. Did it and re-created that last order. It's working.

Popolechien, 2021-12-20:
Well, nope. Both https://cardshop.hotspot.kiwix.org/orders/L765R61C0 and https://cardshop.hotspot.kiwix.org/orders/L766R61C0 crashed again, again with C'est pas sorcier, around 15 and 23% of it.

rgaudin, 2021-12-20:
This is not related to this loop-device ticket at all. Please bump the download-related ticket instead

closed 2021-12-20 by rgaudin

Popolechien, 2021-12-20:
woops my bad ^^