offspot/kiwix-hotspot #631: Erratic network behaviour on Raspberry Pi 3B+

Popolechien, 2022-01-17:
I have created and tested [a small image ](https://cardshop.hotspot.kiwix.org/all-orders/61e1a10798c8bfa4bf65167b) (16Gb) on Raspberry 3B, 3B+, 4 (4Gb), 400 and W (Zero).

* On the 3B I can see and connect to the network without any problem.
* On the 3B+ (tested on two devices), the network either fails to appear, or is seen but cannot be accessed. 
* On RPi4 I cannot see the network 50% of the time. Waiting any amount of time does not change anything. 
* On the W, network is seen and accessed.
* On the 400, network is seen and accessed.

rgaudin, 2022-01-18:
Only testing on HW you reported problems with. Testing with a Pixel 4a on Android 12, macOS Monterey, macOS Big Sur and Windows 10 (VM).

## Pi 4

All OK, seeing and connecting to network `Hello`.

## Pi 3B+

All OK, seeing and connecting to network `Hello`.

--

It's important to note that the network is displayed in the “All networks” section and not listed as a primary network. This is probably because it's an Open network and thus considered less secure.
You could try with an image including a password to check whether it makes the main list. There's no definition as to what should be in that list though. Could be previously seen networks or other criteria.

Given all of this hasn't changed and was working, I am inclined to consider this en environment issue on your end.

kelson42, 2022-10-17:
@Popolechien @rgaudin What shoukd be done next with this. I hardly see what’s going on here.

Popolechien, 2022-10-18:
Haven't seen this happen in a while. Closing ticket.

closed 2022-10-18 by Popolechien