offspot/kiwix-hotspot #636: Failed Process

underthemountain, 2022-02-14:
I have tried this a few different times on different drives. I am not sure what I am having issues with. 
I am trying to create a hotspot on a raspberry pi using my computer to set up the sd card. 
It runs for over 4 hours then fails at 50% every time. I am not sure if I am running out of space on my Temp C? 

Everything seems normal until here: 

[2:25:00 AM] Restoring system sleep policy
[2:25:00 AM] Restoring ES_CONTINUOUS mode to current thread
[2:25:00 AM] *** DURATIONS SUMMARY ***
[2:25:00 AM] [1/6] Retrieve Master Image: 5 minutes and 38.65 seconds (2/13/2022 9:37:05 PM to 2/13/2022 9:42:44 PM)
[2:25:00 AM] [2/6] Download contents: 4 hours, 42 minutes and 15.73 seconds (2/13/2022 9:42:44 PM to 2/14/2022 2:25:00 AM)
[2:25:00 AM] TOTAL: 4 hours, 47 minutes and 54.85 seconds (2/13/2022 9:37:05 PM to 2/14/2022 2:25:00 AM)
[2:25:00 AM] --> Failed: Process ['C:\\Users\\user\\AppData\\Local\\Temp\\
[2:25:00 AM] Installation failed: Process ['C:\\Users\\user\\AppData\\Local\\Temp\\7zSAEC5.tmp\\qemu\\qemu-img.exe', 'resize', '-f', 'raw', 'B:\\kiwix\\hotspot-2022_02_13-21_37_05.BUILDING.img', '274877906944'] failed
[2:25:00 AM] [STAGE 3/6: Image configuration (virtualized) - 50%]

rgaudin, 2022-02-15:
@underthemountain, from this short log extract it seems to be failing during the resize of the file. According to the log, you requested a 256GiB image so yes, it might be that you're running out of space on B: drive.

FYI, on that B:\kiwix folder, the installer will download all contents and, **alongside**, create a file the size you requested your image to be. So if you intended on filling up most of that image, you'd need roughly 512GiB of storage on B: drive to create the image.

That's because we don't download content directly into the image, but on your computer first, then we copy them into the image.

closed 2022-02-15 by rgaudin

underthemountain, 2022-02-16:
That did not work. I am still getting the same error. I have 75GB left on my temp and 217GB left on the drive where the files are being saved.

rgaudin, 2022-02-16:
Was the error exactly the same? 217GB might be too short depending on the exact config and requested size. I'd advise you test on a small (16 or 32GB) config to confirm it's a disk space issue.

underthemountain, 2022-02-16:
When I start my B drive has 381 GB free and my C has 75GB. I dropped down to under half of my B drive space on the latest attempt. 

Here is the new error:

--- Exception Trace ---
Traceback (most recent call last):
  File "run_installation.py", line 292, in run_installation
  File "backend\qemu.py", line 138, in resize_image
  File "backend\util.py", line 104, in subprocess_pretty_check_call
  File "backend\util.py", line 97, in subprocess_pretty_call
backend.util.CheckCallException: Process ['C:\\Users\\user\\AppData\\Local\\Temp\\7zS426F.tmp\\qemu\\qemu-img.exe', 'resize', '-f', 'raw', 'B:\\kiwix\\hotspot-2022_02_15-21_46_01.BUILDING.img', '274877906944'] failed

[1:35:23 AM] --> Failed: Process ['C:\\Users\\user\\AppData\\Local\\Temp\\
[1:35:23 AM] Installation failed: Process ['C:\\Users\\user\\AppData\\Local\\Temp\\7zS426F.tmp\\qemu\\qemu-img.exe', 'resize', '-f', 'raw', 'B:\\kiwix\\hotspot-2022_02_15-21_46_01.BUILDING.img', '274877906944'] failed
[1:35:23 AM] [STAGE 3/6: Image configuration (virtualized) - 50%]
[1:35:23 AM] Restoring system sleep policy
[1:35:23 AM] Restoring ES_CONTINUOUS mode to current thread
[1:35:23 AM] *** DURATIONS SUMMARY ***
[1:35:23 AM] [1/6] Retrieve Master Image: 6 minutes and 8.62 seconds (2/15/2022 9:46:01 PM to 2/15/2022 9:52:09 PM)
[1:35:23 AM] [2/6] Download contents: 3 hours, 43 minutes and 13.37 seconds (2/15/2022 9:52:09 PM to 2/16/2022 1:35:23 AM)
[1:35:23 AM] [3/6] Image configuration (virtualized): 0.43 seconds (2/16/2022 1:35:23 AM to 2/16/2022 1:35:23 AM)
[1:35:23 AM] TOTAL: 3 hours, 49 minutes and 22.45 seconds (2/15/2022 9:46:01 PM to 2/16/2022 1:35:23 AM)

rgaudin, 2022-02-16:
**Again**, you need **twice** the card size free on the disk that hold the _build directory_ (B:\Kiwix).

Here, you try to make a 256GiB card with 381GB of free space. That's not enough. Please, as suggested, confirm with a small test card that the process works

underthemountain, 2022-02-16:
The error above was with 111GB test. I am running on now with 10GB. I will let you know the result.

underthemountain, 2022-02-16:
@rgaudin 
It worked with 10 GB. I guess I will find more space on my hd. Thank you for the help