offspot/kiwix-hotspot #640: Image creatation fails on Ubuntu 21.10

alexbirkett, 2022-02-28:
exfat-fuse, exfat-utils and udisks2 are installed yet I get a `exfat-fuse is required.` error message

[Log](https://pastebin.com/qNRNfPuk)

rgaudin, 2022-03-01:
Thank you for the report. What distribution are you on? This is triggered when `/sbin/mount.exfat` is missing or you don't have execution rights unless you have exfat support in the kernel (visible in `/proc/filesystems`).

rgaudin, 2022-03-01:
Sorry I just saw you mentionned Ubuntu

rgaudin, 2022-03-01:
Can you provide output of `dpkg -l |grep exfat` ?

samuk, 2022-03-26:
@alexbirkett This also just happened to me. It worked after a reboot though. Have you tried it since a reboot?

kelson42, 2022-06-08:
@samuk @alexbirkett Without more information about your respective systems, we will have to close this ticket. Kiwix-hotspot should not use Fuse to mount exfat, this is weird and we need first to understand why.

rgaudin, 2022-06-08:
> Kiwix-hotspot should not use Fuse to mount exfat, this is weird and we need first to understand why.

??? kiwix-hotspot cannot rely on some kernel module or configuration and thus is mostly dependent on exfat-fuse for regular users.

kelson42, 2022-06-08:
@rgaudin This is within the qemu right, so we control it?

kelson42, 2022-06-08:
@rgaudin Thx for correcting my mistake, so this kind of problem might still happen with future image-creator (assuming we stay with exfat).

kelson42, 2022-10-16:
@rgaudin I would propose to close this ticket as `WONTFIX` because we won't rely on `exfat` anymore with the future Kiwix Hotspot 3.0 and this is too much work to fix this ticket for the few next maintenance release of Kiwix-Hotspot 2.

rgaudin, 2022-10-17:
I agree d25db9c

closed 2022-10-17 by rgaudin