offspot/kiwix-hotspot #641: Reduce range of available timezones

Popolechien, 2022-03-15:
The installer offers a large, very large number of possible location for timezones
(here on the desktop)
[image]
(here on the cardshop)
[image]

This is simply too many and poor UX. I remember there was a specific ask that warranted the adding of timezones, but there should be a middle ground that limits itself to CET, EST, IST, etc. (ie 25 segments already)

rgaudin, 2022-03-15:
Agrees this is not convenient but I don't think limiting options is the way to go. There are examples of alternative ways to request this information though.

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42