offspot/kiwix-hotspot #644: Installation failed: Bad CRC-32 for file 'kiwix-hotspot_master_2021-11-24.img'

samuk, 2022-03-26:
Any ideas?

--> Extracting base image from ZIP file
[18:26:24] --> Failed: Bad CRC-32 for file 'kiwix-hotspot_master_2021-11-
[18:26:24] Installation failed: Bad CRC-32 for file 'kiwix-hotspot_master_2021-11-24.img'
[18:26:24] [STAGE 1/6: Retrieve Master Image - 17%]
[18:26:24] 
--- Exception Trace ---
Traceback (most recent call last):
  File "run_installation.py", line 185, in run_installation
  File "backend/download.py", line 269, in unzip_file
  File "zipfile.py", line 1630, in extract
  File "zipfile.py", line 1702, in _extract_member
  File "shutil.py", line 205, in copyfileobj
  File "zipfile.py", line 940, in read
  File "zipfile.py", line 1030, in _read1
  File "zipfile.py", line 958, in _update_crc
zipfile.BadZipFile: Bad CRC-32 for file 'kiwix-hotspot_master_2021-11-24.img'

kelson42, 2022-03-26:
@samuk This looks like somehow the download of the master failed... problem of connection to download it?!

rgaudin, 2022-03-26:
Looks indeed like a download issue. Would you mind retrying?

samuk, 2022-03-26:
It failed in a different way this time:

[19:24:43] 'NoneType' object has no attribute 'groups'
[19:24:43] Traceback (most recent call last):
  File "backend/mount.py", line 248, in test_mount_procedure
  File "backend/mount.py", line 445, in mount_data_partition
  File "backend/mount.py", line 335, in get_virtual_device
AttributeError: 'NoneType' object has no attribute 'groups'

[19:24:43] --> Failed: thorough mount procedure failed
[19:24:43] Installation failed: thorough mount procedure failed
[19:24:43] [STAGE 1/6: Retrieve Master Image - 17%]

samuk, 2022-03-26:
Third attempt: 

[19:29:12] --> Failed: thorough mount procedure failed
[19:29:12] Installation failed: thorough mount procedure failed
[19:29:12] [STAGE 1/6: Retrieve Master Image - 17%]
[19:29:12] Call: ['/usr/bin/xdg-screensaver', 'resume', '0x5200000']
[19:29:13] mv: /tmp/_MEIncW9jQ/libselinux.so.1: no version information available (required by mv)
[19:29:13] --> Restoring loop device (/dev/loop1) mode
[19:29:13] Call: ['pkexec', 'chmod', '-v', '0660', '/dev/loop1']
[19:29:26] mode of '/dev/loop1' changed from 0666 (rw-rw-rw-) to 0660 (rw-rw----)
[19:29:26] *** DURATIONS SUMMARY ***
[19:29:26] [1/6] Retrieve Master Image: 1 minute and 3.33 seconds (Sat 26 Mar 2022 19:28:09  to Sat 26 Mar 2022 19:29:12 )
[19:29:26] TOTAL: 1 minute and 18.49 seconds (Sat 26 Mar 2022 19:27:54  to Sat 26 Mar 2022 19:29:12 )

I asked me for a root pw for Chmod twice this time

samuk, 2022-03-26:
Another attempt:

[22:55:43] Mapped file /home/sam/Desktop/hotspot-2022_03_26-22_54_27.BUILDING.img as /dev/loop2.
[22:55:43] 'NoneType' object has no attribute 'groups'
[22:55:43] Traceback (most recent call last):
  File "backend/mount.py", line 248, in test_mount_procedure
  File "backend/mount.py", line 445, in mount_data_partition
  File "backend/mount.py", line 335, in get_virtual_device
AttributeError: 'NoneType' object has no attribute 'groups'

[22:55:43] 
--- Exception Trace ---
Traceback (most recent call last):
  File "run_installation.py", line 199, in run_installation
ValueError: thorough mount procedure failed

---
[22:55:43] Restoring system sleep policy
[22:55:43] Resuming xdg-screensaver (wid #0x6600000)
[22:55:43] --> Failed: thorough mount procedure failed
[22:55:43] Installation failed: thorough mount procedure failed
[22:55:43] [STAGE 1/6: Retrieve Master Image - 17%]
[22:55:43] Call: ['/usr/bin/xdg-screensaver', 'resume', '0x6600000']
[22:55:43] mv: /tmp/_MEI7ETAZZ/libselinux.so.1: no version information available (required by mv)
[22:55:43] --> Restoring loop device (/dev/loop2) mode
[22:55:43] Call: ['pkexec', 'chmod', '-v', '0660', '/dev/loop2']
[22:55:53] mode of '/dev/loop2' changed from 0666 (rw-rw-rw-) to 0660 (rw-rw----)
[22:55:53] *** DURATIONS SUMMARY ***
[22:55:53] [1/6] Retrieve Master Image: 1 minute and 4.81 seconds (Sat 26 Mar 2022 22:54:38  to Sat 26 Mar 2022 22:55:43 )
[22:55:53] TOTAL: 1 minute and 15.43 seconds (Sat 26 Mar 2022 22:54:27  to Sat 26 Mar 2022 22:55:43 )

kelson42, 2022-03-27:
@rgaudin is the expert, but looks like from the logs that a screensaver is activated. Can you deactivate it?

rgaudin, 2022-03-27:
No it's not related to the screensaver (we disable it during build and restore it at the end) but to the mount procedure : we test it early.

I'll have to look at logs closely but can you tell us what version or the tool you are using and what system you are on ? distrib and version ?

rgaudin, 2022-03-27:
For the record, it's failing at parsing the output of `udisksctl loop-setup` command. I see in your log that the expected output is there (`Mapped file /home/sam/Desktop/hotspot-2022_03_26-22_54_27.BUILDING.img as /dev/loop2.`) and this matches the regex.
As the code is expecting this on the first line of the output, it would be good to know if there's another line reported by udiskctl (even a blank one) before that one.

Since the test fails after a minute, would you mind re-running it and pasting the whole log?

samuk, 2022-03-27:
Sure, 
root@Thinkpad:/home/sam# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"

I used https://download.kiwix.org/release/kiwix-hotspot/kiwix-hotspot-linux.tar.gz

[22:54:27] Getting output of ['/bin/lsblk', '/dev/mmcblk0']
[22:54:27] [STAGE 0/6: Preparations - 0%]
[22:54:27] [STAGE 0/6: Preparations - 0%]
[22:54:27] Preventing system from sleeping
[22:54:27] Suspending xdg-screensaver
[22:54:27] Calling ['/usr/bin/xdg-screensaver', 'suspend', '0x6600000']
[22:54:27] --> Check System Requirements
[22:54:27] Please read https://github.com/kiwix/kiwix-hotspot/wiki/System-Requirements for details
[22:54:27] --> Ensure user files are present
[22:54:27] --> Prepare Image file
[22:54:27] Call: ['/usr/bin/udisksctl', 'dump']
[22:54:27] /usr/bin/udisksctl: /tmp/_MEI7ETAZZ/libselinux.so.1: no version information available (required by /lib/x86_64-linux-gnu/libmount.so.1)
[22:54:27] /org/freedesktop/UDisks2/Manager:
[22:54:27]   org.freedesktop.UDisks2.Manager:
[22:54:27]     DefaultEncryptionType:      luks1
[22:54:27]     SupportedEncryptionTypes:   luks1
[22:54:27]                                 luks2
[22:54:27]     SupportedFilesystems:       ext2
[22:54:27]                                 ext3
[22:54:27]                                 ext4
[22:54:27]                                 vfat
[22:54:27]                                 ntfs
[22:54:27]                                 exfat
[22:54:27]                                 xfs
[22:54:27]                                 reiserfs
[22:54:27]                                 nilfs2
[22:54:27]                                 btrfs
[22:54:27]                                 minix
[22:54:27]                                 udf
[22:54:27]                                 f2fs
[22:54:27]                                 swap
[22:54:27]     Version:                    2.9.2
[22:54:27] 
[22:54:27] /org/freedesktop/UDisks2/block_devices/loop0:
[22:54:27]   org.freedesktop.UDisks2.Block:
[22:54:27]     Configuration:              []
[22:54:27]     CryptoBackingDevice:        '/'
[22:54:27]     Device:                     /dev/loop0
[22:54:27]     DeviceNumber:               1792
[22:54:27]     Drive:                      '/'
[22:54:27]     HintAuto:                   false
[22:54:27]     HintIconName:               
[22:54:27]     HintIgnore:                 true
[22:54:27]     HintName:                   
[22:54:27]     HintPartitionable:          true
[22:54:27]     HintSymbolicIconName:       
[22:54:27]     HintSystem:                 true
[22:54:27]     Id:                         
[22:54:27]     IdLabel:                    
[22:54:27]     IdType:                     exfat
[22:54:27]     IdUUID:                     E1E4-8DFB
[22:54:27]     IdUsage:                    filesystem
[22:54:27]     IdVersion:                  1.0
[22:54:27]     MDRaid:                     '/'
[22:54:27]     MDRaidMember:               '/'
[22:54:27]     PreferredDevice:            /dev/loop0
[22:54:27]     ReadOnly:                   false
[22:54:27]     Size:                       1585740288
[22:54:27]     Symlinks:                   /dev/disk/by-uuid/E1E4-8DFB
[22:54:27]     UserspaceMountOptions:      
[22:54:27]   org.freedesktop.UDisks2.Filesystem:
[22:54:27]     MountPoints:        
[22:54:27]     Size:               0
[22:54:27]   org.freedesktop.UDisks2.Loop:
[... 602 lines cut ...]
  File "backend/mount.py", line 248, in test_mount_procedure
  File "backend/mount.py", line 445, in mount_data_partition
  File "backend/mount.py", line 335, in get_virtual_device
AttributeError: 'NoneType' object has no attribute 'groups'

[22:55:43] 
--- Exception Trace ---
Traceback (most recent call last):
  File "run_installation.py", line 199, in run_installation
ValueError: thorough mount procedure failed

---
[22:55:43] Restoring system sleep policy
[22:55:43] Resuming xdg-screensaver (wid #0x6600000)
[22:55:43] --> Failed: thorough mount procedure failed
[22:55:43] Installation failed: thorough mount procedure failed
[22:55:43] [STAGE 1/6: Retrieve Master Image - 17%]
[22:55:43] Call: ['/usr/bin/xdg-screensaver', 'resume', '0x6600000']
[22:55:43] mv: /tmp/_MEI7ETAZZ/libselinux.so.1: no version information available (required by mv)
[22:55:43] --> Restoring loop device (/dev/loop2) mode
[22:55:43] Call: ['pkexec', 'chmod', '-v', '0660', '/dev/loop2']
[22:55:53] mode of '/dev/loop2' changed from 0666 (rw-rw-rw-) to 0660 (rw-rw----)
[22:55:53] *** DURATIONS SUMMARY ***
[22:55:53] [1/6] Retrieve Master Image: 1 minute and 4.81 seconds (Sat 26 Mar 2022 22:54:38  to Sat 26 Mar 2022 22:55:43 )
[22:55:53] TOTAL: 1 minute and 15.43 seconds (Sat 26 Mar 2022 22:54:27  to Sat 26 Mar 2022 22:55:43 )

samuk, 2022-03-29:
I've actually taken a different approach now, so resolving this may not be worth your time. Feel free to re-open if you do want to poke at it.

rgaudin, 2022-03-30:
Sorry about the delay ; as you can see in your log, udiskctl reports an error before output the expected message. That's what's causing the issue. We can fix this as expecting first line is fragile but that would be for next release.

closed 2024-06-04 by kelson42