offspot/kiwix-hotspot #645: Open hardware support?

samuk, 2022-03-28:
Is there any interest in supporting open hardware? Perhaps the [Radxa Zero ](https://wiki.radxa.com/Zero/hardware/zero)

Popolechien, 2022-03-28:
@samuk Oh we have plenty of interest, but no developer available unfortunately. But overall and considering the current shortage having alternatives to RPi can't be a bad idea.

kelson42, 2022-03-28:
Would be nice, but for the moment totally out of question because of lack of resources :( Maintaining the solution on all versions of RPI is already a burden.

samuk, 2022-03-28:
OK thanks, I'll explore whether a homebrew version is viable, perhaps https://github.com/coova/coova-chilli or similar

samuk, 2022-03-28:
Or maybe https://piratebox.cc/raspberry_pi:diy:armbian

closed 2022-03-28 by samuk