offspot/kiwix-hotspot #650: Hotspot fails to connect more than 15 tablets simultaneously

Popolechien, 2022-06-03:
Orange's supplier tried to connect their tablets to the hotspot (specifically with [this image](https://cardshop.hotspot.kiwix.org/orders/L1055R6289)) and they say they can only connect 15 devices simultaneously.

The tablets are running Android 10, I'm also attaching the user manual
[FP-TAB-V2.1-pact.pdf](https://github.com/offspot/kiwix-hotspot/files/8832270/FP-TAB-V2.1-pact.pdf)
[FP-TAB-V2.2-pact.pdf](https://github.com/offspot/kiwix-hotspot/files/8832272/FP-TAB-V2.2-pact.pdf)

Any idea what went wrong?

Popolechien, 2022-06-03:
I'll confirm the scenario but they're running RPi 3B+. The first 15 tablets connect to the device, but from 16 onwards they can't establish a connection to the server.

rgaudin, 2022-06-03:
Not sure what this means ; please confirm with them ; thanks

kelson42, 2022-08-09:
@rgaudin Looks like we have an explanation and a fix ;)

commit 382a940b (2022-08-09): Fixed #650: dont limit number of wifi clients
closed 2022-08-09 by rgaudin