offspot/kiwix-hotspot #653: Add drivers for long range RPi antennas

Popolechien, 2022-06-27:
Orange's provider tested their Raspberry Pis over the week-end and it appears that none of the long-range antennas they added to the set up work or even are recognized.
They're asking that we add extra drivers to cover for each antenna.

rgaudin, 2022-06-27:
Can you get the drivers or packages for those drivers ?

kelson42, 2022-06-27:
What is the problem we try to solve here basically? Long range antenna, why not... but if the wifi client are "normal", not sure this will help.

Popolechien, 2022-06-27:
> Can you get the drivers or packages for those drivers ?

I've sent a note to that effect.

kelson42, 2022-10-17:
@rgaudin @Popolechien Now that we have proper limit, I guess this ticket can be closed as WONTFIX?

rgaudin, 2022-10-18:
Yes, reported hasn't even sent feedback regarding the fix. Given the need was unclear and the amount of support it requires, this is wontfix for now.

closed 2022-10-18 by rgaudin