offspot/kiwix-hotspot #654: Asks for login and password on bootup - Default Free Image for PI

FluxTux, 2022-07-21:
I completed the online install of default (free) package for Raspberry Pi. Used imager to flash a 128 gig card, booted in Pi.

Obvious errors are FAILED to start dhcpd on all interfaces > on board and 2 x usb WIFi work using Pi OS by default

Then I am asked for login and password, which I cannot seem to find anywhere online or in forum.

THanks for help

rgaudin, 2022-07-21:
The hotspot is mostly meant as a plug and connect via WiFi device (hence the name).
You're free to use it as you please though of course.

the dhcpd error line is normal and displays on all boots.

Use of additional network cards is not supported but you're free to tweak the system for it to work.

Credentials are the same as the raspbian it is based off:
username: pi 
password: raspberry

closed 2022-07-21 by rgaudin

Popolechien, 2022-07-21:
@FluxTux A bit more clearly: you got this prompt because you plugged your Pi onto a screen. But this will not be of any use because the only way to access the content is to connect via wifi.

FluxTux, 2022-07-21:
Thank you for your responses