offspot/kiwix-hotspot #660: Unable to create an image for Raspberry pi, various errors

kelson42, 2022-08-22:
From overview created by [scottmayo](https://github.com/scottmayo): kiwix/overview#72  

First, the hotspot installer app doesn't provide a vertical scrollbar if it doesn't fit on the screen - I had to plug in an external monitor just to use it on my laptop. Also, it prompts repeatedly for the root password - if it wants to be run sudo it should demand that once, up front.

Second, selecting sites to fill a 256G SD card created a .img that would not fit on the card, even though the hotspot builder said I had space left as I was selecting packages.

So I trimmed down the cache and tried again. I let the app wipe the SD card, and...

[08:54:37 AM] Getting output of ['/bin/lsblk', '/dev/mmcblk0']
[08:54:37 AM] [STAGE 0/6: Preparations - 0%]
[08:54:37 AM] [STAGE 0/6: Preparations - 0%]
[08:54:37 AM] Preventing system from sleeping
[08:54:37 AM] Suspending xdg-screensaver
[08:54:37 AM] Calling ['/usr/bin/xdg-screensaver', 'suspend', '0x6800000']
[08:54:37 AM] --> Check System Requirements
[08:54:37 AM] Please read https://github.com/kiwix/kiwix-hotspot/wiki/System-Requirements for details
[08:54:37 AM] --> Ensure user files are present
[08:54:37 AM] --> Prepare Image file
[08:54:37 AM] Call: ['/usr/bin/udisksctl', 'dump']
[08:54:37 AM] /org/freedesktop/UDisks2/Manager:
[08:54:37 AM]   org.freedesktop.UDisks2.Manager:
[08:54:37 AM]     SupportedFilesystems:       ext2
[08:54:37 AM]                                 ext3
[08:54:37 AM]                                 ext4
[08:54:37 AM]                                 vfat
[08:54:37 AM]                                 ntfs
[08:54:37 AM]                                 exfat
[08:54:37 AM]                                 xfs
[08:54:37 AM]                                 reiserfs
[08:54:37 AM]                                 nilfs2
[08:54:37 AM]                                 btrfs
[08:54:37 AM]                                 minix
[08:54:37 AM]                                 udf
[08:54:37 AM]                                 f2fs
[08:54:37 AM]                                 swap
[08:54:37 AM]     Version:                    2.7.6
[08:54:37 AM] 
[08:54:37 AM] /org/freedesktop/UDisks2/block_devices/loop0:
[08:54:37 AM]   org.freedesktop.UDisks2.Block:
[08:54:37 AM]     Configuration:              []
[08:54:37 AM]     CryptoBackingDevice:        '/'
[08:54:37 AM]     Device:                     /dev/loop0
[08:54:37 AM]     DeviceNumber:               1792
[08:54:37 AM]     Drive:                      '/'
[08:54:37 AM]     HintAuto:                   false
[08:54:37 AM]     HintIconName:               
[08:54:37 AM]     HintIgnore:                 false
[08:54:37 AM]     HintName:                   
[08:54:37 AM]     HintPartitionable:          true
[08:54:37 AM]     HintSymbolicIconName:       
[08:54:37 AM]     HintSystem:                 true
[08:54:37 AM]     Id:                         
[08:54:37 AM]     IdLabel:                    
[08:54:37 AM]     IdType:                     
[08:54:37 AM]     IdUUID:                     
[08:54:37 AM]     IdUsage:                    
[08:54:37 AM]     IdVersion:                  
[08:54:37 AM]     MDRaid:                     '/'
[08:54:37 AM]     MDRaidMember:               '/'
[08:54:37 AM]     PreferredDevice:            /dev/loop0
[08:54:37 AM]     ReadOnly:                   true
[08:54:37 AM]     Size:                       274877906944
[08:54:37 AM]     Symlinks:                   
[08:54:37 AM]     UserspaceMountOptions:      
[08:54:37 AM]   org.freedesktop.UDisks2.Loop:
[08:54:37 AM]     Autoclear:          true
[08:54:37 AM]     BackingFile:        /home/scott/Documents/hotspot-2022_08_19-22_34_38.ERROR.img (deleted)
[08:54:37 AM]     SetupByUID:         0
[08:54:37 AM]   org.freedesktop.UDisks2.PartitionTable:
[08:54:37 AM]     Partitions:         ['/org/freedesktop/UDisks2/block_devices/loop0p3', '/org/freedesktop/UDisks2/block_devices/loop0p2', '/org/freedesktop/UDisks2/block_devices/loop0p1']
[... 998 lines cut ...]
[08:56:37 AM] Error creating textual authentication agent: Error opening current controlling terminal for the process (`/dev/tty'): No such device or address (polkit-error-quark, 0)
[08:56:37 AM] Error deleting loop device /dev/loop2: GDBus.Error:org.freedesktop.UDisks2.Error.NotAuthorizedCanObtain: Not authorized to perform operation
[08:56:37 AM] 
--- Exception Trace ---
Traceback (most recent call last):
  File "run_installation.py", line 199, in run_installation
ValueError: thorough mount procedure failed

---
[08:56:37 AM] Restoring system sleep policy
[08:56:37 AM] Resuming xdg-screensaver (wid #0x6800000)
[08:56:37 AM] Call: ['/usr/bin/xdg-screensaver', 'resume', '0x6800000']
[08:56:37 AM] --> Restoring loop device (/dev/loop1) mode
[08:56:37 AM] Call: ['pkexec', 'chmod', '-v', '0660', '/dev/loop1']
[08:56:37 AM] --> Failed: thorough mount procedure failed
[08:56:37 AM] Installation failed: thorough mount procedure failed
[08:56:37 AM] [STAGE 1/6: Retrieve Master Image - 17%]
[08:56:47 AM] mode of '/dev/loop1' changed from 0666 (rw-rw-rw-) to 0660 (rw-rw----)
[08:56:47 AM] *** DURATIONS SUMMARY ***
[08:56:47 AM] [1/6] Retrieve Master Image: 1 minute and 48.69 seconds (Sun 21 Aug 2022 08:54:49 AM  to Sun 21 Aug 2022 08:56:37 AM )
[08:56:47 AM] TOTAL: 2 minutes and 0.78 seconds (Sun 21 Aug 2022 08:54:37 AM  to Sun 21 Aug 2022 08:56:37 AM )

No idea what this means, but this is a quite generic (if older) install of Linux Mint and mounting things should not be a challenge.

If this had worked at all well I would have considered contributing money and perhaps time to the project, because I feel like the basic idea is worthwhile. But it doesn't run on Pop-Os (exfat_utils not available), doesn't run on laptops (non-resizable window doesn't fit on screen), doesn't calculate sizes correctly and then fails incomprehensibly in ways that suggest the SD card isn't set up right, even after it's formatted to be clean and even when the app gets to wipe the SD itself.

rgaudin, 2022-08-30:
The size comment is unclear as it mixes SD card size and cache…

It looks like attaching the virtual device had it mounted automatically. This is detected via the text output of the udisk2 command we use. Might be that this output is different on this Mint version than the Ubuntu one we tested with.

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42