offspot/kiwix-hotspot #661: Installer fails but first wastes an hour of the user's time

kelson42, 2022-08-22:
From overview created by [scottmayo](https://github.com/scottmayo): kiwix/overview#70  

Run the installer and answer a lot of questions. Spend an hour researching and selecting static content to fill a 256G SD.
Run installation. 
It fails because it cannot find exfat-utils, and I don't even know why it needs it - the SD is already clean. When you choose abort, the whole application exits, and _none of your settings are saved_.

Check your preconditions BEFORE you waste an hour of the user's time sorting through a disorganized list of hundreds of topics. 

Also, there would be a way to build an image from .zim files I've already downloaded, and it would be way nicer if you could copy zim files onto the boot disk and have the server find them. Having to build them into the image makes no sense and causes a lot of effort if you just want to add something.

kelson42, 2022-08-22:
I guess this is a ticket for kiwix-hotspot repository.

kelson42, 2022-08-22:
@rgaudin I wonder indeed if it would not be a good idea to check things earlier? In particular, we are going to need this I guess, even after the revamp?

rgaudin, 2022-08-22:
It's like this on "purpose", because kiwix-hotspot is not only an image creator but a configuration creator: one can spend an hour selecting content and settings and save this configuration to a file for later use.

With the revamp, we should be able to have an informative message about the requirements

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42