offspot/kiwix-hotspot #662: Hotspot bootloader will not boot on RasberryPi 3b

c3bud, 2022-08-29:
Used Windows installer program, downloaded all img files and flashed to brand new clean 128gb sd card, this created three partitions, one for the boot, one for all the files and then a leftover empty partition.

I turn off my working RaspberryPi, change out the sd card and put the hotspot card in, and turn back on. I get an error on boot every time.
Here is the error message:
end kernel panic - not syncing: VFS: unable to mount root fs on unknown block (179.3)
3.742122

Suggestions?

rgaudin, 2022-08-30:
Looks like the third partition (179.**3**) is corrupted and needs a filesystem check. Since that partition is of exFAT format, you can try to insert it into your computer, ask windows to check it (third partition only!) and retry.

Another alternative is to reflash the image to the SD-card. We recommend using https://github.com/raspberrypi/rpi-imager/releases

c3bud, 2022-08-30:
I will try your suggestions. Not sure why the original flashing with etcher created three partitions.
Partition 1 - boot. Etcher flashed it to Fat32 and 252 MB capacity, using 47.6 MB leaving 204 MB of free space
Partition 2 - HSPOT_DATA. Etcher flashed it to keep exFAT with 121 GB capacity, using 105 GB for the files and leaving 16.4 GB of free space
Partition 3 - USB Drive. Etcher kept this as exFAT with 6.26 GB capacity, with 192 KB used and 6.26 GB of free space, but drive says that it is empty. I have reformatted this partition using the Windows default Format from a right click in Explorer.

If boot still fails, I will try your second suggestion of reflashing the card. However do I just reformat each partition before doing that or is there a way to remove the partitions and recreate the single partition of 128 GB? I've already ruined a different brand new 128 GB SD card by using the "wipe" from the Kiwix installer program, it turned 128 GB card into a 100 MB card and for the life of me I could not reformat it back.

rgaudin, 2022-08-31:
@c3bud, the installer creates those 3 partitions on purpose:
- p1 is `boot` (fat32)
- p2 is system (extfs). On windows, this would be seen as unknown and it's unreadable.
- p3 is `HSPOT_DATA` (exFAT). That's where all your selected content is stored.

Since you said you formatted one (or more) of the partitions, it's useless to try as either your formatted the system one and it won't boot or the data one and it will have no data package (you mixed the order in your list).

You should reflash the card with the image. You don't need to reformat anything, the flashing tool (use the one I suggested above) will take care of what needs to be done.

The *wipe* feature is meant as a previous step before flashing a new image. If you want to use the card without an image, you need to create a partition on it using the Windows Disk utility for instance.

c3bud, 2022-09-05:
The reflash with the one you recommended above did allow the boot, however I did not set an Admin username or password so could not log in.

I started over from the beginning with a brand new download of the files and created a new image. This time using the default username of "admin" but changing the password. I used a variety of numbers and upper and lower case letters. No other symbols. The download did not show any errors that I am aware of, the image file flashed to the sd card just fine. The boot loader worked to asking for the username, it accepted the username, but would never accept the password. 

What could I possibly be doing wrong? It seems like such a simple process. I do have to run my download and image creation again because I realized I had a space in the hotspot name, which will not work for the URL. But I don't want to do this until someone can give me a clue on the Admin username and password issue.

Thanks in advance for additional assistance.

rgaudin, 2022-09-05:
admin username and password are for the apps that have an admin mode (khan, edupi, aflatoun, wikifundi). The hotspot is targeting WiFi-use, not display-use.

If you want to log-in via the console, when plugged into a screen, the username is `pi` and the password `raspberry` but nothing awaiting you there, except the raspbian console.

closed 2022-09-05 by rgaudin