offspot/kiwix-hotspot #663: Add Gcompris

Popolechien, 2022-08-30:
Note: this is a software package (android/macOS/Linux/Windows) so not exactly for a zim file but could be integrated to the hotspot in the same way as other apps?

- Website URL: https://www.gcompris.net/index-en.html
- License: **CC-by-sa **
- Desired ZIM Title: **Gcompris**
- Desired ZIM Description: **GCompris is a high quality educational software suite, including a large number of activities for children aged 2 to 10. Some of the activities are game orientated, but nonetheless still educational.**
- Desired ZIM Icon –png (URL or attach one): https://abc.com/icon.png
- Language (ISO 639-3): **fra/eng**

rgaudin, 2022-08-30:
Could but should be discussed as well as it shifts focus slightly. Offspot is mainly a web platform and we integrated those *apps* for a single user. Was only android app at first, then a Windows version of some was included alongside the android one – do we know if there are any users for those?

Adding Gcompris ourselves would open the door for becoming a software catalog which I don't think we want.

I am more in favor of including a better edupi replacement with an easy to use initial-package that users would build. This way, if they want Gcompris but don't want to include Linux version (which package system would we provide anyway?) they can simply add URL/files to the Windows/mac ones.

Popolechien, 2022-08-30:
Yeah this needs to be discussed but I wanted it logged somewhere and the alternative was zim-requests.

At the end of the day we're helping the spread of educational material, so this is relevant.

rgaudin, 2022-08-30:
Sure is relevant. Current behavior is too high maintenance and poor user experience though. There are multiple ways to approach this.