offspot/kiwix-hotspot #664: Adding local ZIM files

rgaudin, 2022-08-30:
From #661, in addition to to-come OPDS library+download, having the ability to add locally stored ZIM files onto the image would be very useful.

This is different than the cache which stores the in-OPDS ZIM file which can be superseded at any moment.

kelson42, 2022-10-16:
@rgaudin Do we plan to allow this in Kiwix Hotspot `3.0.0`? Maybe we should move the ticket somewhere else?

Popolechien, 2022-10-17:
I reckon that enabling this possibility earlier rather than later would make sense if we want to encourage a possible integration with zimit.

rgaudin, 2022-10-17:
For `3.0.0`, probably not but it's a good feature so for a later release, that makes sense.

We'll have to keep it here until we're at a point we have a repo for the new version (assuming we're switching repo). I dislike empty repos with issues: the process is not settled and it's hard to turn them into actionable tickets afterwards. So I'd leave them here until then.

kelson42, 2024-06-04:
AFAIK this has been implement in Kiwix Hotspot 3.0 which propose a way to upload a ZIM file via a web form/

closed 2024-06-04 by kelson42