offspot/kiwix-hotspot #666: UI hardly readable on small display

rgaudin, 2022-08-30:
UI doesn't scale down and requires ~ 460x800px. On screens with less than this available, this means that the window is not fully visible.

A vertical scroll on the window would help but a _responsive_ UI would be better obviously.

scottmayo, 2022-08-31:
Disagree that this is an enhancement. The tool is completely unusable on laptops as is.

rgaudin, 2022-08-31:
Can you provide your laptop resolution? We all work on laptops

rgaudin, 2022-08-31:
Classified as enhancement because the UI will be redone soon as its been terrible and hard to maintain for years but you're right

scottmayo, 2022-08-31:
Can I make a plea for a redesign in general?

I think the basic idea is wonderful; I really  want to be able to access 
offline data from a raspberry pi. But you don't have a working solution.

1) You don't have a local-only solution for a raspberry pi. I want to be 
able to browse .zlm files on a pi without it having to be a wireless 
access point. And there's no reason why even a pi zero can't run a 
browser against a local-only (no network needed) server that reads .zlm. 
It will be slow, but in emergency situations slow but power efficient 
might be much more acceptable than the high power draw of a laptop. 
While I like the idea of a pi as an access point with a web server, 
there should be an option for the pi that doesn't emit wifi at all and 
just lets me browse.

2) Your code seems to rely on a host of shell based applications to do 
everything. These vary between platforms and you clearly don't test on 
the same versions of linux that I use (mint, pop_os). You need to stick 
to POSIX calls and apps.

3) Why do you insist on writing a bootable image for raspberry pis? Pis 
already have bootable images and all you need to do is reconfigure them 
into wifi access points and load in a web server that reads  your .zlm 
files. You shouldn't be in the business of reformatting disks and 
writing boot images. Write a script that takes an existing SD for a pi 
and adds to it what you need. You'll go from relying on handfuls of 
fragile apps to needing nothing more than fopen. Best of all, that 
leaves people with a pi that still runs other things if they really need 
that.

I think access to life-critical information in an offline fashion is a 
fantastic idea - and I look forward to the day when it works.

rgaudin, 2022-08-31:
Well a whole redesign is in progress but what you seem to be looking for is [kiwix-serve](https://download.kiwix.org/release/kiwix-tools/): a simple command-line tool that serves ZIM content over HTTP.
Just download it (it's a static binary) and launch it, passing it the ZIM files you have. Up to you to do this on a running graphical session or as a service on your hotspot-serving Pi…

this project has very different constraints and users. I think you landed on the wrong repo. Feel free to suggest anything though but please, open separate tickets for those as multiple-topics tickets are harder to work with.

kelson42, 2022-10-30:
@scottmayo After reading this ticket, I still don’t understand what is exactly the problem. If a rendering is not OK, please upload a screenshot and clearly explain what you expect. A new ticket shoukd be open, this ticket is mixing too many different things.

closed 2022-10-30 by kelson42