offspot/kiwix-hotspot #668: Add Survival Manual

Popolechien, 2022-09-14:
https://github.com/ligi/SurvivalManual is a fully Libre / GPL3 app with tons of survival tips, available in several languages.

kelson42, 2022-10-16:
@rgaudin I believe we should move the app requests to `container-images` and create a dedicated tag `new_app`

rgaudin, 2022-10-17:
> @rgaudin I believe we should move the app requests to `container-images` and move the tag `application` there too.

I agree, but I'd propose two use status tags like `requested-app` and `accepted-app` (rejected are closed as `wontfix`) so that we know we need to implement accepted ones and not-settled ones stands out. Otherwise we'll just reproduce what we had: tickets in the bin that stays there forever.

That said, this ticket is not about an app. It's a file-sharing package candidate and those have not been extensively discussed yet.

Popolechien, 2022-10-17:
>  this ticket is not about an app. It's a file-sharing package candidate and those have not been extensively discussed yet.

What do you mean exactly? I have the app installed on my phone. Isn't the git repo hosting it?

rgaudin, 2022-10-17:
It's not an _Offspot App_, it's an Android App. As much as offspot is concerned, this is a file to offer for download (file-sharing), just like nomad.

kelson42, 2022-10-17:
> @rgaudin I believe we should move the app requests to container-images and move the tag application there too.\n\nI agree, but I'd propose two use status tags like requested-app and accepted-app (rejected are closed as wontfix) so that we know we need to implement accepted ones and not-settled ones stands out. Otherwise we'll just reproduce what we had: tickets in the bin that stays there forever.

It is important indeed to know if a feature request is "accepted"... or not yet. This is a general problem/requirement with the tickets. Not only a problem with this kind of requests.

My approach so far was to put the "question" tag if I was not sur if the feature request should be implemented or not. I propose to do the dame here... and I'm sorry if this approach was so far unclear.

rgaudin, 2022-10-17:
The problem is not with the approach, which was clear but the process. We need a process to discuss and settle feature-requests in a timely manner and close rejected one.