offspot/kiwix-hotspot #670: Move directory "online-demo" to a dedicated repository

kelson42, 2022-10-30:
This is important we keep the ability to run Kiwix hotspot on an x86 standalone server without wifi (so without captive portal)... etc.

The current documentation as been made and is helpful but will have to be revamped once we will move to Docker based solution.

Anyway, to me this usage has nothing to do with kiwix-desktop repository anymore.

Therefore I propose to move the current document to a new repository, for example `offspot/kiwix-hotspot-cloud`

rgaudin, 2022-10-30:
There is close to nothing to keep from that doc as it was very specific to qemu and the previous code base.
It's not possible to write the new doc yet because the new tool isn't available but it will be 95% deployment specific so it's important we (you and @Popolechien ) lay out the goal and specifications of that demo-tool as it will be differently implemented depending on what we want to do with it.

closed 2022-10-30 by kelson42

kelson42, 2022-11-05:
See `offspot/dashboard` for the future versions...

rgaudin, 2022-11-05:
Not sure I see the connection between the online demo and the dashboard 🤔