offspot/kiwix-hotspot #671: UI unusable on small display

scottmayo, 2022-10-30:
Without a vertical scrollbar there is no way to get to much of the window on small displays. The application becomes unusable.

A major point of the software is that you can get to information in emergencies, and in emergencies all that might be available are low power, battery powered devices. Low power devices generally have tiny displays - 320x480 might be all that's available. If the application supported scroll bars, it would at least be possible to use it. Right now it is not possible.

(Let me suggest that the ideal use case for emergencies situations is running some sort of GUI directly on a raspberry pi zero with a display as small as 320x480, with no need for a network. This would be a standalone app, not a server; in an emergency you don't want to have to run a separate server and client just to see your information, or waste battery on a wifi. It would be slow, but that doesn't matter.)

kelson42, 2022-10-30:
@scottmayo Please upload screenshot of the page which has a problem

rgaudin, 2022-10-30:
Problem is already identified and discussed in a separate ticket. Kiwix Hotspot 2 UI is not responsive and has no vertical scrollbar so it's not usable on tiny screens because the start button is at the very bottom.

Hotspot UI though, the one running on the Pi, have always been responsive.

kelson42, 2022-10-30:
So, this ticket is really to complain the the installer can not be displayed properly on 320x480?!? Sorry, but to me this is a `wontfix`. Makes no sense.

scottmayo, 2022-10-30:
Sorry, I'm submitted on the wrong kiwix product.

The problem is that I've abandoned all attempts to use kiwix - it wouldn't install on my older copy of POP_OS (too many dependencies), it didn't show a usable GUI on another laptop because the screen was small, and I don't want the server solution because that would involve powering both a raspberry pi and a separate computer to view things and I don't want to emit wifi just to read pages. I'm not going to try to reinstall, so I'm not in a position to provide screen shots.

Basically I'm asking for something to be developed that lets me view offline copies of wikipedia and other sites on a single, low power raspberry pi with a small screen. Do whatever you want with the request as I've already put too many hours into trying to install and use it and won't be contributing further effort.

kelson42, 2022-10-30:
> I don't want the server solution because that would involve powering both a raspberry pi and a separate computer to view things and I don't want to emit wifi just to read pages.

All of this is wrong: you can have a server (kiwix-serve HTTP daemon) and reading on the same machine via a browser.

After all these message I'm still unsure about which software you really talk... now it seems to be kiwix-desktop. Kiwix-desktop will have difficulties to display on 320x480 which is indeed super small... but if you have a browser which runs, then reading from kiwix-serve will be doable.... assuming the content of the ZIM can resize properly.

Here, the core problem is that the bug report was not focusing enough describing the problem and more on advising what to do.... at the end nobody here is able to help properly. Usually we know what to do once we fully understand a problem, but otherwise there is no way out.

closed 2022-10-30 by kelson42