offspot/kiwix-hotspot #674: Failure to Download base image

iatethecookies, 2023-04-16:
I'm very new to Kiwix and tinkering with the Raspberry Pi in general, so I don't know how to troubleshoot effectively. I have been following guides, but when I try to create and image for my SD card, I am left with a message that says there has been a "Failure to download base image." I'm not too sure how to fix this so help would be appreciated.

[7:37:23 PM] Getting output of ['wmic', 'partition']
[7:37:23 PM] [STAGE 1/6: Retrieve Master Image - 0%]
[7:37:23 PM] [STAGE 1/6: Retrieve Master Image - 0%]
[7:37:23 PM] Preventing system from sleeping
[7:37:23 PM] Setting ES_SYSTEM_REQUIRED mode to current thread
[7:37:23 PM] --> Check System Requirements
[7:37:23 PM] Please read https://github.com/kiwix/kiwix-hotspot/wiki/System-Requirements for details
[7:37:23 PM] --> Ensure user files are present
[7:37:23 PM] --> Prepare Image file
[7:37:23 PM] --> Dumping Hotspot Configuration
[7:37:23 PM] {
    "disk_size": 127829076480,
    "wikifundi_languages": [],
    "packages": [
        "wikipedia_en_all_maxi.en"
    ],
    "aflatoun_languages": [],
    "mathews": false,
    "catalogs": [
        {
            "url": "http://mirror.download.kiwix.org/library/ideascube.yml",
            "name": "Kiwix",
            "description": "Kiwix ZIM Content",
            "local_url": "file:///home/pi/catalog_Kiwix.yml"
        }
    ],
    "kalite_languages": [],
    "admin_password": "****",
    "kernel_version": "4.19.57",
    "timezone": "America/Los_Angeles",
    "root_partition_size": 7000000000,
    "language_name": "English",
    "project_name": "wiki",
    "mirror": "http://download.kiwix.org",
    "language": "en",
    "nomad": false,
    "admin_account": "admin",
    "custom_branding_path": "/tmp",
    "installer_version": "2.3.6",
    "edupi_has_resources": false,
    "edupi": false
}
[7:37:23 PM] homepage saved to: C:\Users\Austi\AppData\Local\Temp\tmpk3ry260i.html
[7:37:23 PM] [STAGE 1/6: Retrieve Master Image - 0%]
[7:37:23 PM] --> Retrieving base image file
[7:37:23 PM] C:\Users\Austi\AppData\Local\Temp\7zS6BD4.tmp\aria2c.exe --dir=A:\Kiwix\kiwix-hotspot.cache --out=hotspot-master_2019-08-24.img.zip --connect-timeout=60 --max-file-not-found=5 --max-tries=5 --retry-wait=60 --timeout=60 --follow-metalink=mem --allow-overwrite=true --always-resume=false --max-resume-failure-tries=1 --auto-file-renaming=false --download-result=full --log-level=error --console-log-level=error --summary-interval=1 --human-readable=false --ca-certificate=C:\Users\Austi\AppData\Local\Temp\7zS6BD4.tmp\ca-certificates.crt http://download.kiwix.org/hotspots/base/hotspot-master_2019-08-24.img.zip
[7:37:23 PM] [                                                          ] 0%
[7:37:24 PM] [..........................................................] 100%

[7:37:24 PM] 
[7:37:24 PM] Failed to download base image.
aria2c returned 4
[7:37:24 PM] Restoring system sleep policy
[7:37:24 PM] Restoring ES_CONTINUOUS mode to current thread
[7:37:24 PM] *** DURATIONS SUMMARY ***
[7:37:24 PM] TOTAL: 1.25 second (4/15/2023 7:37:23 PM to 4/15/2023 7:37:24 PM)

kelson42, 2023-04-16:
@iatethecookies Can you please here share the details (with logs) of what you do and get?

kelson42, 2023-04-16:
User has decided to empty his message on its own. I guess this is because he has solved his problem.

closed 2023-04-16 by kelson42