offspot/kiwix-hotspot #675: Unable to launch installer because SSL issue

kelson42, 2023-04-18:
I downloaded the windows installer on the "Kiwix for Raspberry Pi" page, however when I try to run the "kiwix-hotspot-win64-2.4.12" file on my Windows 11 machine, I receive the following error:
"
Failed to execute script 'main' due to unhandled exception: OpenSSL 3.0's legacy provider failed to load. This is a fatal error by default, but cryptography supports running without legacy algorithms by setting the environment variable CRYPTOGRAPHY_OPENSSL_NO_LEGACY. If you did not expect this error, you have likely made a mistake with your OpenSSL configuration.
"

Open first at https://github.com/kiwix/kiwix-desktop/issues/933

kelson42, 2023-04-18:
@Moordok314 what is your exact version of Windows?

rgaudin, 2023-04-18:
Confirmed on Windows 11.

[image]

@Moordok314, we won't be releasing a fixed version very soon so you'll have to defer to this hack:

- open windows menu and type `environ` 
[image]

- Open the *Edit the system environment variables* app
- Click the *Environment Variables* button at the bottom of the window
- Click the *New* button on the upper pane
- Type (**exactly**) `CRYPTOGRAPHY_OPENSSL_NO_LEGACY` as *Variable Name*
- Type `1` as *Variable value` and hit *OK*. Your window should look like:

[image]

Close the other window with *OK*. You can now launch kiwix-hotspot.

closed 2023-04-18 by rgaudin