offspot/kiwix-hotspot #677: Hotspot Installer Content Catalog Fail

kelson42, 2023-05-08:
From kiwix-desktop created by [mikejfeser](https://github.com/mikejfeser): kiwix/kiwix-desktop#941  

[image]
[image]

Hello,

When running the hotspot installer, I am unable to load static content. It warns that I should check internet/proxy settings, however, when going to proxy settings and test connection, I can confirm that I can reach the Kiwix servers. Any assistance would be much appreciated!

kelson42, 2023-05-08:
This is a ticket for https://github.com/offspot/kiwix-hotspot, I will move it.

rgaudin, 2023-05-09:
I can't reproduce on Win11. Which version of Windows are you running?
Can you access `http://mirror.download.kiwix.org/library/ideascube.yml` from your web browser?

mikejfeser, 2023-05-09:
Windows 10, and yes I can access that address.

rgaudin, 2023-06-06:
I'm sorry I can't reproduce the issue.

[image]

It shouldn't make a difference but I'd advise you to try the 2.4.11 as 2.4.12 has an important bug on Windows that would affect you anyway.

LuxStatera, 2023-09-28:
Same issue on Linux and Mac

LuxStatera, 2023-10-21:
I just tried this link "http://mirror.download.kiwix.org/library/ideascube.yml" and it works. It opens a page with a bunch of text. I'm in the US

rgaudin, 2023-10-23:
I'll look into it. Are you using a Mac? You seem to have tested on various platforms but I'll try to reproduce on yours

LuxStatera, 2023-10-23:
I am using a Mac but I also tried Linux. Both failed. Also I can't find the Windows version so I wasn't able to test on Windows. The Kiwix website only shows a Mac and Linux installer

LuxStatera, 2023-12-06:
@rgaudin Also do you have a link to the Windows installer? I haven't tried that one yet because I couldn't find it

LuxStatera, 2023-12-10:
@rgaudin Thank you. I tried version 2.4.11 and 2.4.13 on Windows 10 but I am still getting the same error

LuxStatera, 2024-02-09:
@rgaudin How did the testing go?

rgaudin, 2024-02-09:
Sorry I could not find time yet to look at this. Thanks for the reminder.

LuxStatera, 2024-04-10:
Not stale. 

@rgaudin what's the word?

rgaudin, 2024-04-10:
Sorry I'm running out of ideas. Couple things to try:

- Running the GUI from the Console so we get a chance to see if there's additional debug info. There might be a line like `Exception while downloading/parsing catalogs` with details. To do this, you must unpack the windows exe file with 7zip then launch it from the console/terminal. On macOS it's simple as `/Applications/Kiwix\ Hotspot.app/Contents/MacOS/kiwix-hotspot` from the terminal. Beware launching from terminal doesn't focus the App Window. Find it in dock to reveal it.
- Running the CLI version. It's not embedded in Windows version I think so that would be macOS or Linux. The most reliable version, anyway, is Linux version with `--root` flag.

The CLI version can be used with a JSON Config file created with the GUI. That's handy so you don't have to use CLI to select content which is hardly for human… but for that you'd need the GUI to be able to access the catalog.

```sh
kiwix-hotspot cli --root --config myconfig.json --size 16G
```

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42