offspot/kiwix-hotspot #678: I can´t execute in W10 Pro 1909 18363.1256

runikk, 2023-05-19:
Hi, 

I tried to execute the .exe version 2.4.12 downloaded from the following link: https://www.kiwix.org/en/documentation/how-to-set-up-kiwix-hotspot/  in 3 different computers with aproximately the same SSL error in all of them:

[image]

I don´t know what is happening. I also tried Debian in Virtualbox and at 50% aprox a SSH error with connection appears and a message of "error installation".

rgaudin, 2023-05-20:
Duplicate of #675. Can you try the workaround described there ?

runikk, 2023-05-20:
Done! #675. SSL solved but... now I have a different error
[image]

rgaudin, 2023-05-20:
Ah! We are indeed trying to decode UTF-8 while reading the output of a Windows tool that lists the disk drives. Output is thus system-encoded and I suppose it can fail if any drive have a caption that is not overlapping.

kelson42, 2024-06-04:
Unfortunately this product has been discontinued. Therefore this issue won't be fixed/implemented. A bit more details can be found [here](https://github.com/offspot/kiwix-hotspot/blob/main/README.md).

closed 2024-06-04 by kelson42