offspot/kiwix-hotspot #680: module 'os' has no attribute 'sync' #944

rgaudin, 2023-06-01:
_Originally reported by @Moordok314 in https://github.com/kiwix/kiwix-desktop/issues/944_

After the last zim file copies in stage 4/6 of the instillation, I receive the following error and then the instillation terminates. I am running windows 11 pro if that helps.

[image]

rgaudin, 2023-06-01:
Indeed this line should only be called on Unix platftorms, not Windows

Moordok314, 2023-06-01:
> ZjQcmQRYFpfptBannerStart
> This Message Is From an External Sender
> This message came from outside your organization.
> ZjQcmQRYFpfptBannerEnd

So how do I fix that? I'm using the "kiwix-hotspot-win64-2.4.12" installer
for the Raspberry Pi hotspot.

rgaudin, 2023-06-02:
There's no way for you to work around it unfortunately. You'd have to wait for us to fix.

If you can't wait, you can run the software on Linux (or macOS):

- It's not very convenient but you can install https://www.virtualbox.org/ and use one of the Linux images at [osboxes.org](https://www.osboxes.org/ubuntu/).
- Alternatively, you can export your configuration from the kiwix-hotspot version you are using and use the linux version (command-line only) via [Windows Subsystem Linux](https://learn.microsoft.com/en-us/windows/wsl/install)

rgaudin, 2023-06-06:
Another simpler alternative might be to revert to v2.4.11. Most notable changes are:

- Updated WikiFundi
- Updated Africatik
- Lifted a limit on the number of simultaneous WiFi clients

closed 2023-06-08 by rgaudin